neuropsychology/NeuroKit #413: Creation of Uneven Epochs

jonchuasoonkiat, 2021-02-10:
Hi friends,

Thank you for creating such an amazing kit for easy analysis of data.

I was just wondering how we might be able to create epochs that are uneven for the time-related analysis of data. I noticed a note 

I propose to include a variable in the source code for the function epochs_create() named "event_duration" that accesses the "duration" array created by the events_find() function.

Subsequently, we can assign the end of the event as event_start + "duration". 

I'm not an expert on coding but your team might be able to make this work. Hope to see this capability appear soon!

Thanks a bunch!

Tam-Pham, 2021-02-11:
Hi @jonchuasoonkiat,

Glad you enjoy using NeuroKit and find it useful for your work :)

Even though with a different workflow from the one you suggested, the package does support creating uneven epochs.  `epochs_start` and `epochs_end` arguments in `epochs_create()` can be input as lists. For instance, in the example below, I would like to create 5 epochs with the same start time (relative to `events_onset` from events) but each with a different end time. I can do the following:

```python
epochs_end = [41, 20, 33, 23, 39]
epochs = nk.epochs_create(data, events, epochs_start=0, epochs_end=epochs_end)
``` 
Hope it helps :)

DominiqueMakowski, 2021-02-11:
that said, could we add a possibility of setting `epochs_end="from_events"` that would indeed pull the duration key from the events?

closed 2021-03-10 by Tam-Pham