neuropsychology/NeuroKit #416: Butterworth filter not working

MeenaSiddharthan, 2021-02-25:
Hi!! So I can't run nk.ecg_process to get cleaned ECG signals. 
So this is my line:
ecg_signals, ecg_info = nk.ecg_process(data["ECGBIT"], sampling_rate=sampling_rate)

And this is the error I got:
Traceback (most recent call last):
.
.
.
  File "C:\Users\m.siddharthan\AppData\Local\Continuum\anaconda3\lib\site-packages\neurokit2\signal\signal_filter.py", line 211, in _signal_filter_butterworth
    sos = scipy.signal.butter(order, freqs, btype=filter_type, output="sos", fs=sampling_rate)

TypeError: butter() got an unexpected keyword argument 'fs'

Is there somewhitng I'm missing? But it seemed to work to get events/get plots etc

Tam-Pham, 2021-02-25:
Could you check what's your scipy version? Because fs is a new argument that is just added in version 1.2.0

DominiqueMakowski, 2021-02-25:
@MeenaSiddharthan you can get the versions by running `nk.version()`

[image]

MeenaSiddharthan, 2021-02-25:
Heyyy! Thanks! So here are my versions:
- OS: Windows (WindowsPE 64bit) 
- Python: 3.7.1 
- NeuroKit2: 0.1.1 
- NumPy: 1.15.4 
- Pandas: 0.23.4 
- SciPy: 1.1.0 
- sklearn: 0.20.1 
- matplotlib: 3.0.2

JanCBrammer, 2021-02-25:
@MeenaSiddharthan, please upgrade your SciPy to version 1.2.0. The `fs` argument to `scipy.signal.butter` did not exist prior to the 1.2.0 version (see https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.butter.html).

Tam-Pham, 2021-02-25:
Ah! Seems like your scipy is not up to date. Ya can update it to the latest and try again :)

MeenaSiddharthan, 2021-02-26:
Yup! I upgraded scipy now everything works! Thanksss

closed 2021-02-26 by DominiqueMakowski