neuropsychology/NeuroKit #418: Fix nk.standardize() to exclude data that are strings

Tam-Pham, 2021-03-02:
Potentially have it check for data.dtypes. TODO

DominiqueMakowski, 2021-03-02:
we should add an `is_string()` helper (that works with vectors and dataframes - in which case it returns a boolean array that we can use to index the df); this will probably come in handy in the future

DominiqueMakowski, 2021-03-02:
Also, we should add an `nk.find_outliers()` function like the one from neurokit 1 :

```python
def find_outliers(data, treshold=2.58):
    """
    Identify outliers (abnormal values) using the standart deviation.

    Parameters
    ----------
    data : list or ndarray
        Data array
    treshold : float
        Maximum deviation (in terms of standart deviation). Rule of thumb of a gaussian distribution: 2.58 = rejecting 1%, 2.33 = rejecting 2%, 1.96 = 5% and 1.28 = rejecting 10%.

    Returns
    ----------
    outliers : ndarray
        A list of True/False with True being the outliers.

    Example
    ----------
    >>> import neurokit as nk
    >>> outliers = nk.find_outliers([1, 2, 1, 5, 666, 4, 1 ,3, 5])
    """
    outliers = []
    mean = np.mean(data)
    std = np.std(data)
    for i in data:
        if abs(i - mean)/std < treshold:
            outliers.append(False)
        else:
            outliers.append(True)
    outliers = np.array(outliers)
    return outliers
```

But also allow users to insert not the threshold SD, but the cumulative proportion, like `[0.025, 0.975]` that would exclude the 5% extreme observations.

DominiqueMakowski, 2021-03-02:
- [ ] Also, when the data is only nans, avoid throwing a warning

closed 2021-03-10 by Tam-Pham