neuropsychology/NeuroKit #419: What is the unit of frequency domain variable (LF,HF) in HRV analysis

Wan-umaidah, 2021-03-04:
**Question and context**
I have computed HRV in frequency domain. I have wondering that why the  value of LF, HF in hrv dataframe  is very low compared to PSD (ms2/Hz) plot. So I have question that what is the unit of LF and HF?????. Please let me know.

Another question, Cloud I customize the color, font and so on in Time/frequency/Non-linear plot in HRV ??

Tam-Pham, 2021-03-05:
Hi @Wan-umaidah, 

The unit of LF and HF in the dataframe depends on the unit of your input. If your r-peaks series is in seconds, the unit of the power output will be s2/Hz. If the input is in milliseconds, the units of the power output will be ms2/Hz. 

>  I have wondering that why the value of LF, HF in hrv dataframe is very low compared to PSD (ms2/Hz) plot.

Could you share your plot and your dataframe here to help us understand the issue better?

> Another question, Cloud I customize the color, font and so on in Time/frequency/Non-linear plot in HRV ??

You can do so by customizing your processing pipeline. Please refer to the example of how to do so [here](https://neurokit2.readthedocs.io/en/latest/examples/custom.html)

Wan-umaidah, 2021-03-05:
I input in raw EEG (mv). So the unit of LF or HF is mv2/Hz right? but in the frequency plot is ms2/Hz
[image] Did I do anything wrong?

[NTV9.xlsx](https://github.com/neuropsychology/NeuroKit/files/6088405/NTV9.xlsx) >> raw ECG in mv
[n_ntv9.xlsx](https://github.com/neuropsychology/NeuroKit/files/6088415/n_ntv9.xlsx) >> out put from neurokit2

Wan-umaidah, 2021-03-08:
Please let me know

Tam-Pham, 2021-03-08:
Hi @Wan-umaidah 

Sorry for the late reply. Please don't mind the label on the plot. We currently have it hardcoded to ms2/Hz but it does not reflect the actual unit of the frequency. We have it in such a way as the input in the default processing pipeline is R-R intervals, which is in ms. Will look into adjusting it to unit2/Hz later to be more general.

Could you share your entire processing script with me? Just to make sure that the input is correct.

Wan-umaidah, 2021-03-08:
> Hi @Wan-umaidah
>
> Sorry for the late reply. Please don't mind the label on the plot. We currently have it hardcoded to ms2/Hz but it does not reflect the actual unit of the frequency. We have it in [...]

@Hi Tam-Pham
This is my script to compute HRV
[image]
[image]
[image]

Wan-umaidah, 2021-03-09:
Is there anything wrong??

DominiqueMakowski, 2021-03-09:
As explained above, the name of the axis could be misleading. We will double-check it, but we cannot give you any date or timeline (note that there's no need of adding new comments every day: as long as the issue is open, we know it is something we need to look at - when we have time).

That said, if you are in a hurry, you can always help us looking through the code of how the density is computed. After all, this is what open-source is for.

Tam-Pham, 2021-03-09:
Hi @Wan-umaidah 

I have taken a look at your code and seems like everything is alright.

> I have wondering that why the value of LF, HF in hrv dataframe is very low compared to PSD (ms2/Hz) plot

The power value that you see in the hrv dataframe is obtained by integrating the spectral density (the area under the curve). You can read more about it [here](https://skill-lync.com/projects/File-Parsing-and-Data-Analysis-in-Python-Part-II-Engine-Performance-82072). The values might appear very small because the default option in `hrv_frequency()` for normalization is `True`. Set it to `False` and you will obtain relatively much larger values. 

> I input in raw EEG (mv). So the unit of LF or HF is mv2/Hz right? but in the frequency plot is ms2/Hz

Please note that even though the unit for raw ECG is mv, the input to calculating different frequency domain indices is the R-R intervals, which is in ms or s. If you are using the default pipeline of NeuroKit, as shown in your screen captures, ms2 is the correct value as the R-R intervals are obtained by:

https://github.com/neuropsychology/NeuroKit/blob/25bc913b95a4bf4e162c64cc96a40d084c0337c5/neurokit2/hrv/hrv_utils.py#L10

Hope we have answered your questions.

closed 2021-03-10 by Tam-Pham