neuropsychology/NeuroKit #420: Events identified wrongly

MeenaSiddharthan, 2021-03-15:
Hi there~

I have this strange problem with my more recent recordings where when I try to find events based on LUX,
>2 events are returned.

So my line to find events are as follows:
>events = nk.events_find(data["LUX"], threshold_keep='below', event_conditions=["Spontaneous", "Deep Breathing"])

And my error message is as follows:
>ValueError: NeuroKit error: events_find(): oops, it seems like you provided 209203 `event_conditions`, but 209203 events got detected :(. Check your event conditions or the event signal!

But when I plot the recorded signals, there are only 2 distinct time windows there LUX falls below zero. 
[image]

Is the signal not clean enough? I'm not sure why there are so many events detected.

```Versions used:
- OS: Windows (WindowsPE 64bit) 
- Python: 3.7.1 
- NeuroKit2: 0.1.1 

- NumPy: 1.20.1 
- Pandas: 1.2.2 
- SciPy: 1.6.1 
- sklearn: 0.20.1 
- matplotlib: 3.0.2
```

zen-juen, 2021-03-15:
Hi @MeenaSiddharthan 😄 

How many events are you expecting to detect? You can try to additionally specify `threshold=0`  in the function if let's say you want to keep only events where LUX falls below 0. It might be because the default value for `threshold` is between the max and min of the LUX values that is giving you more events than you expect.

Tam-Pham, 2021-03-15:
Btw, @MeenaSiddharthan can you try to update NeuroKit to the latest version of 0.1.1? We updated the error message for events_find() to reflect the accurate mismatch in the number of events provided and detected. Here the message is showing the same number 😅

MeenaSiddharthan, 2021-03-15:
@zen-juen Hey heyyy

So I tried doing that and I get like the same issue but with lesser number of events at least.
So I changed my line to:
>events = nk.events_find(data["LUX"], threshold=0,threshold_keep='below', event_conditions=["Spontaneous", "Deep Breathing"])

But got the error message:
> ValueError: NeuroKit error: events_find(): oops, it seems like you provided 46427 `event_conditions`, but 46427 events got detected :(. Check your event conditions or the event signal!

Then I tried 
> events = nk.events_find(data["LUX"], threshold=-0.5,threshold_keep='below', event_conditions=["Spontaneous", "Deep Breathing"])

And I didn't get any error messages but I wasn't able to extract the times of the two events as arrays either. 

@Tam-Pham I can try! Is it just pip update neurokit?

zen-juen, 2021-03-15:
Hmm that's weird, it seems like your neurokit version is the latest already and the changes we made in `events_find()` are not reflected in the error message - can you check if it's indeed neurokit2 0.1.1? If not can you `pip install neurokit2==0.1.1`? Then try with `threshold=-1` if you are keeping only events below -1

MeenaSiddharthan, 2021-03-15:
@zen-juen Yeahhh I tried installing neurokit2 0.1.1 again using the line you sent me. I still get the many many events.

I tried threshold=-1
>events = nk.events_find(data["LUX"], threshold=-1,threshold_keep='below', event_conditions=["Spontaneous", "Deep Breathing"])

But I get this message:
>C:\Users\m.siddharthan\AppData\Local\Continuum\anaconda3\lib\site-packages\neurokit2\events\events_find.py:94: NeuroKitWarning: No events found. Check your event_channel or adjust 'threshold' or 'keep' arguments.
  category=NeuroKitWarning

Hmmmmmm I don't get ittttttt

MeenaSiddharthan, 2021-03-18:
Thanksssss @zen-juen  for telling me about duration_min!!!

closed 2021-03-18 by MeenaSiddharthan