neuropsychology/NeuroKit #421: Creating an ndarray from ragged nested sequences is deprecated

yywangvr, 2021-03-15:
Thanks for the open source Neurokit2 package.

**Describe the bug**
Frequently, I get the the following warning. Maybe that Neurokit2 needs updating? 
```bash
 miniconda3/envs/py38/lib/python3.7/site-packages/neurokit2/eda/eda_findpeaks.py:289: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray
  scrs = np.array(scrs)
```

**System Specifications**

```python
 import neurokit2 as nk
nk.version()
- OS: Darwin ( 64bit) 
- Python: 3.7.10 
- NeuroKit2: 0.0.42 

- NumPy: 1.19.2 
- Pandas: 1.2.3 
- SciPy: 1.6.1 
- sklearn: 0.24.1 
- matplotlib: 3.3.4
```

yywangvr, 2021-03-16:
In eda_findpeaks.py, I replaced 

```python
scrs = np.array(scrs)
```
with
```python
scrs = np.array(scrs, dtype=object)
```
Then, I bypassed the warning...However, I am not sure this will affect the final results...

JanCBrammer, 2021-03-16:
@yywangvr, reopening since this is something we might want to address in the source code. Thanks for reporting!

Tam-Pham, 2021-03-19:
Thanks @yywangvr for opening the issue with the warning. 

@JanCBrammer agree! This warning surfaces due to the update in the latest [Numpy 1.19.0 update](https://numpy.org/doc/1.19/release/1.19.0-notes.html#deprecate-automatic-dtype-object-for-ragged-input) regarding "ragged input". 

The `scrs`  is returned is a list of lists with uneven lengths and `dtype=object` is not the default anymore. Numpy requires us to specify in the function. 

https://github.com/neuropsychology/NeuroKit/blob/883bbb3461d1e40dc5f16a1f6eb2a7babb9f0758/neurokit2/eda/eda_findpeaks.py#L289

However, upon looking at the script, I'm not sure why do we have to perform this step 😅 since object `scrs` is not returned anyway @DominiqueMakowski

JanCBrammer, 2021-03-19:
> object `scrs` is not returned anyway

@Tam-Pham, turning `srcs` into a NumPy array seems unnecessary, yes.

closed 2021-04-04 by Tam-Pham