neuropsychology/NeuroKit #425: bio_analyze(): Allow for different window lengths for different signals

DominiqueMakowski, 2021-03-18:
We want longer epochs for instance for EDA (around 7 s after onset), than for heart rate short-term changes (around 4s) that for EMG. One way is to take the longest and then use subsets for shortest.

zen-juen, 2021-03-19:
We can add an argument `window_lengths` in `nk.bio_analyze()` specifying the subsets, with the specific epoch start and end times for different signals such as: {'ECG': [0.5, 1.5], 'EDA': [0.5, 3.5]} 

This internal function should do the trick:
```
def _bio_analyze_slicewindow(data, window_lengths, signal='ECG'):

    if signal in window_lengths.keys():
        start = window_lengths[signal][0]
        end = window_lengths[signal][1]
        epochs = {}
        for i, label in enumerate(data):
            # Slice window
            epoch = data[label].loc[(data[label].index > start) & (data[label].index < end)]
            epochs[label] = epoch
            epochs
    return epochs

```
which we then call it in `nk.bio_analyze()` to subset the epochs just before the analysis is performed:
```
    if len(ecg_cols) != 0:
        ecg_data = data.copy()
        
        if window_lengths != 'constant':
            if 'ECG' in window_lengths.keys():  # only for epochs
                ecg_data = _bio_analyze_slicewindow(ecg_data, window_lengths, signal='ECG')

        ecg_analyzed = ecg_analyze(ecg_data, sampling_rate=sampling_rate, method=method)
        features = pd.concat([features, ecg_analyzed], axis=1, sort=False)
```

DominiqueMakowski, 2021-08-20:
closing this as I think it's implemented (@zen-juen)

closed 2021-08-20 by DominiqueMakowski