neuropsychology/NeuroKit #426: Why do RR intervals and BB intervals have *1000 in their calculation?

bananakiu, 2021-03-18:
The way I understand RR intervals in ECG is that they are time elapsed between two successive R-waves, while the way I understand BB intervals in RSP is that they are time elapsed between inhalation peaks. Given this, why are the calculations for rri and bbi the following?

RRI (found in NeuroKit/neurokit2/hrv/hrv_utils.py line 10)
rri = np.diff(peaks) / sampling_rate * 1000

BBI (found in NeuroKit/neurokit2/rsp/rsp_rrv.py line 87)
bbi = np.diff(peaks) / sampling_rate * 1000

I am particularly confused about why the * 1000 is necessary.

zen-juen, 2021-03-19:
Hi @bananakiu this is to convert the intervals to milliseconds!

bananakiu, 2021-03-19:
> Hi @bananakiu this is to convert the intervals to milliseconds!

Thanks! That didn't occur to me.

closed 2021-03-19 by bananakiu