neuropsychology/NeuroKit #428: Calculating time of inspiration and expiration using RSP

Mitchellb16, 2021-03-19:
Hi, I had the idea to include average inspiration and expiration times in the output of rsp_intervalrelated, but my solution right now seems a bit unpythonic (is that a word?). Any ideas for making this cleaner? 

```rsp = nk.rsp_simulate(duration=600, sampling_rate=250, respiratory_rate=15)

# Process it
signals, info = nk.rsp_process(rsp, sampling_rate=250)

# Get indexes of phase changes
phase_peaks = signals[signals.RSP_Phase_Completion == 1]

# Calc diff for each phase change
phase_peaks['phase_len'] = np.diff(phase_peaks.index, prepend=0) 
phase_peaks['phase_time'] = phase_peaks.phase_len/250

# make df for expirations and inspirations
insp = phase_peaks[phase_peaks.RSP_Phase == 0]
exp = phase_peaks[phase_peaks.RSP_Phase == 1]

print("Mean insp_time:", insp.phase_time.mean())
print("Mean exp_time:", exp.phase_time.mean())

DominiqueMakowski, 2021-03-22:
we actually recently thought about something related I think right @Tam-Pham?

zen-juen, 2021-03-24:
This will definitely be quite useful as seen from papers describing how expiration/inspiration ratio affects other physiological indexes
- https://journals.physiology.org/doi/full/10.1152/japplphysiol.00163.2013
- https://www.researchgate.net/profile/Maximilian-Moser-2/publication/12418712_Relative_Timing_Of_Inspiration_And_Expiration_Affects_Respiratory_Sinus_Arrhythmia/links/59e492a7a6fdcc7154e1190e/Relative-Timing-Of-Inspiration-And-Expiration-Affects-Respiratory-Sinus-Arrhythmia.pdf

We could perhaps extract the ratio of expiration time to inspiration time (E/I-ratio)?

DominiqueMakowski, 2021-03-24:
'Tis a set of indices we should add to rsp_intervalrealted right?

zen-juen, 2021-03-24:
Yes, seems like the E/I ratio is measured mostly during paced breathing which lasts for a few minutes.

On a related note, we might also want to include an index of heart rate asymmetry (HSA) which seems to be affected by the E/I ratio (as @Tam-Pham mentioned before! see [this paper](https://iopscience.iop.org/article/10.1088/0967-3334/33/10/1717/meta)) 

Calculation of HRA found [here](https://www.researchgate.net/profile/Przemyslaw-Guzik-2/publication/6734042_Heart_rate_asymmetry_by_Poincare_plots_of_RR_intervals/links/00463516712a5287a9000000/Heart-rate-asymmetry-by-Poincare-plots-of-RR-intervals.pdf) 😄 perhaps can open a new issue for this

DominiqueMakowski, 2021-03-24:
Ok let's add these then I think we can directly open a new PR for these RSP & ECG asymmetry indices

zen-juen, 2021-03-24:
Oops, we actually do have HRA indices already 😆 let's just open a new one for the E/I ratio

Mitchellb16, 2021-03-24:
@zen-juen Would the E/I ratio be in addition to average E and I times? I'd be interested in having access to these values for calculating inspiratory flow rate (used in [this paper](https://journals.physiology.org/doi/full/10.1152/japplphysiol.00500.2001)).

DominiqueMakowski, 2021-03-25:
Yeah I think we can add both

zen-juen, 2021-03-25:
@Mitchellb16 I've just added the average inspiration and expiration times as well as the I/E ratio into the output of `rsp_intervalrelated()` - could you install the dev version and see if this fits what you want?

Mitchellb16, 2021-03-25:
@zen-juen sure! Is there a means installing the dev version with pip or would it involve cloning the repo locally?

zen-juen, 2021-03-26:
@Mitchellb16 you can run `pip install https://github.com/neuropsychology/neurokit/zipball/dev` 😄

Mitchellb16, 2021-03-26:
Thanks :) I seem to be getting negative expiration time values for some, but not all of my files. My guess is this might have to do with the checking for unequal lengths step? In that not all exp_end values are ending up after (in time) their respective exp_start values.

Mitchellb16, 2021-03-26:
Ah never mind, I believe this is on my end. I was running a function to remove artifact rows from the rsp_signals df before feeding it into rsp_intervalrelated which totally does not work for this calculation! Though this begs a different question of when the best step in the pipeline is best for removing artifacts. I don't want them contributing to the averages calculated by rsp_intervalrelated, but removing them before rsp_process would totally mess with temporal information.

zen-juen, 2021-03-27:
Yes the pipeline in `rsp_process()` already includes cleaning the signal and fixing and formatting the peaks so having an extra function to remove artifacts in signals might be *cleaning* it twice.

It will be interesting though to see how neurokit's base functions fare in comparison to the function that you have for artifacts removal! 😄

Mitchellb16, 2021-03-27:
My function is rather rudimentary, just some thresholding for rate and amplitude values that we've determined to be outside of what we expect to be physiologically possible. I get my signal from whole-body plethysmography in animals, which opens itself up to large changes in the signal caused by movement, not breathing.

zen-juen, 2021-03-31:
@Mitchellb16 ah thanks for raising this! Let me fix this in a bit and get back to you 😄

zen-juen, 2021-03-31:
Just fixed @Mitchellb16 😄 definitely something I forgot to consider, thanks for pointing it out!

Mitchellb16, 2021-04-01:
Okay I've made my way into where my anaconda files are and it looks like that pip command didn't update my local nk package (which is weird because it worked the first time for the first insp/exp update). I made the changes manually and it fixed it. Sorry for the trouble!

commit 28e6d1ea (2021-03-31): minor fix

#428
closed 2021-04-01 by Mitchellb16