neuropsychology/NeuroKit #430: Technique in CWT delineation

artur-deluca, 2021-03-26:
**Question and context**
Hi, y'all! A quick question that I wasn't able to find in the documentation: what's the technique used to delineate the ECG using the continuous wavelet transform. Does somebody have a publication to point out to me?

Thanks a lot :)

Tam-Pham, 2021-03-26:
hi @artur-deluca 

Essentially both the discrete and continuous wavelet methods were following [Martínez (2004)](https://ieeexplore.ieee.org/document/1275572). We used the continuous wavelet method at the beginning just because it was easier to implement. However, following the paper closely, the discrete wavelet method is the more accurate and robust option. Hope that clarifies

artur-deluca, 2021-03-26:
It does! Thanks a lot @Tam-Pham

closed 2021-03-26 by artur-deluca

DominiqueMakowski, 2021-03-26:
@Tam-Pham maybe we might want to add a bit more details in the docs