neuropsychology/NeuroKit #431: error while running the code on python to extract the R peaks from the signal using ecg_peaks() and calculate the heart period using ecg_rate().

labakoum, 2021-03-27:
Salut tout le monde , je veux tester et Comprendre NeuroKit , j'ai commencer d'afficher un signale ECG de meme base de donnée qui deja existe sur la page du site neuroKit (https://neurokit2.readthedocs.io/en/latest/examples/edr.htm)

Jusqu'a maintenant ca marche bien jusqu'a maintenant , mais lorsque j'ai applique le code qui extraire les pics R et calcule la  période cardiaque. le code ca marche pas et il me une erreur , des idée s'il veut plus les amis et merci .

the database link : **https://raw.githubusercontent.com/neuropsychology/NeuroKit/dev/data/ecg_1000hz.csv**
the original link of the program on neurolink : **https://neurokit2.readthedocs.io/en/latest/examples/edr.htm**

here is now my code which the same code in the link above with the error 

# Load NeuroKit and other useful packages
import neurokit2 as nk
import numpy as np
import pandas as pd
import matplotlib.pyplot as plt
%matplotlib inline
plt.rcParams['figure.figsize'] = [15, 5]  # Bigger images
​
# Get data
ecg = pd.read_csv("https://raw.githubusercontent.com/neuropsychology/NeuroKit/dev/data/ecg_1000hz.csv")
# Visualize signal
nk.signal_plot(ecg)

# Extract peaks
rpeaks, info = nk.ecg_peaks(ecg, sampling_rate=1000)
​
# Compute rate
ecg_rate = nk.ecg_rate(rpeaks, sampling_rate=1000, desired_length=len(ecg))
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
~/opt/anaconda3/lib/python3.8/site-packages/pandas/core/indexes/base.py in get_loc(self, key, method, tolerance)
   2894             try:
-> 2895                 return self._engine.get_loc(casted_key)
   2896             except KeyError as err:

pandas/_libs/index.pyx in pandas._libs.index.IndexEngine.get_loc()

pandas/_libs/index.pyx in pandas._libs.index.IndexEngine.get_loc()

pandas/_libs/hashtable_class_helper.pxi in pandas._libs.hashtable.PyObjectHashTable.get_item()

pandas/_libs/hashtable_class_helper.pxi in pandas._libs.hashtable.PyObjectHashTable.get_item()

KeyError: 'ECG_Clean'

The above exception was the direct cause of the following exception:

KeyError                                  Traceback (most recent call last)
<ipython-input-11-27e4ceda617b> in <module>
      1 # Extract peaks
----> 2 rpeaks, info = nk.ecg_peaks(ecg, sampling_rate=1000)
      3 
      4 # Compute rate
      5 ecg_rate = nk.ecg_rate(rpeaks, sampling_rate=1000, desired_length=len(ecg))

~/opt/anaconda3/lib/python3.8/site-packages/neurokit2/ecg/ecg_peaks.py in ecg_peaks(ecg_cleaned, sampling_rate, method, correct_artifacts)
     69 
     70     """
---> 71     rpeaks = ecg_findpeaks(ecg_cleaned, sampling_rate=sampling_rate, method=method)
     72 
     73     if correct_artifacts:

~/opt/anaconda3/lib/python3.8/site-packages/neurokit2/ecg/ecg_findpeaks.py in ecg_findpeaks(ecg_cleaned, sampling_rate, method, show)
    121     if isinstance(ecg_cleaned, pd.DataFrame):
    122         try:
--> 123             ecg_cleaned = ecg_cleaned["ECG_Clean"]
    124         except NameError:
    125             try:

~/opt/anaconda3/lib/python3.8/site-packages/pandas/core/frame.py in __getitem__(self, key)
   2900             if self.columns.nlevels > 1:
   2901                 return self._getitem_multilevel(key)
-> 2902             indexer = self.columns.get_loc(key)
   2903             if is_integer(indexer):
   2904                 indexer = [indexer]

~/opt/anaconda3/lib/python3.8/site-packages/pandas/core/indexes/base.py in get_loc(self, key, method, tolerance)
   2895                 return self._engine.get_loc(casted_key)
   2896             except KeyError as err:
-> 2897                 raise KeyError(key) from err
   2898 
   2899         if tolerance is not None:

KeyError: 'ECG_Clean'

zen-juen, 2021-03-30:
Hi there @labakoum! The error is indicating that an 'ECG_Clean' column is not found in the signal as our `ecg_peaks()` function takes a cleaned ECG signal rather than a raw one. 

```
ecg_cleaned = nk.ecg_clean(ecg, sampling_rate=1000)
```
All you'll need is to clean the signal first (like the line above) before passing the cleaned output into `ecg_peaks()` and `ecg_rate()`!

labakoum, 2021-03-30:
Hi there @zen-juen  thank you a lot for your answer , I want to try to do this step and then I want to leave you an answer here

closed 2021-04-19 by zen-juen