neuropsychology/NeuroKit #433: Inter-beat intervals and nk.hrv

marianoalbaladejo, 2021-03-29:
Hi,

I have a sensor that returns in a list the inter-beat intervals (distance in milliseconds between peaks) and not the PPG original signal.  I want to use the hrv function to get the heart rate metrics, but my problem is that I can not use the hrv function because it requires a peak list. Is there a function to convert the inter-beat intervals to a peak list or another way to get the heart rate measures from the inter-beat intervals?

Thanks for your help.

DominiqueMakowski, 2021-03-30:
Hi @marianoalbaladejo unless there's something I missed, you can retrieve the peaks location by a cumulative sum of the IBI:

```python
import numpy as np

ibi = [500, 400, 700, 500, 300, 800, 500]
peaks = np.cumsum(ibi)
peaks
```
```
array([ 500,  900, 1600, 2100, 2400, 3200, 3700], dtype=int32)
```

Let me know if that helps!

marianoalbaladejo, 2021-03-30:
I have seen in the documentation that the nk.hrv input is a dictionary returned by ecg_findpeaks, ecg_peaks, ppg_findpeaks, or ppg_peaks. My problem is that I have the IBI and not the PPG signal. For this reason, I can't use the functions that generate the required dictionary. Is it correct to use the cumulative sum of the IBI as input for the nk.hrv method? 

Thanks so much for the help.

marianoalbaladejo, 2021-03-31:
I have done different experiments, and I have got the same results with the first peak (0) and without it. Thanks for the solution! 

But finally, I have a little question, what is the sampling rate for an IBI sequence? Because I have an IBI each time that a peak is detected. I understand the concept of sampling rate for signals, but not for the distance between the peaks of a signal because it changes. The BVP sensor measure with a frequency of 64 Hz, but if I use it in the nk.hrv, the results have no sense.  I have an IBI for every peak and not with constant frequency.  I have made some tests and compared the results with the results obtained by another device, and I get that If the IBI is in milliseconds, the correct sample rate is 1000, and if IBI is in seconds (IBI/1000), the correct sampling rate es 1.  Does this conclusion have an explication due to how the nk.hrv function works? 

I am very grateful for the received answers and the interest in the problem. Thanks to everyone!

DominiqueMakowski, 2021-04-01:
> The BVP sensor measure with a frequency of 64 Hz

This is indeed most likely unrelated to how the IBIs are expressed. If it's Human heart rates, you can probably infer the correct "sampling rate" (in this case, the unit of the IBI) by computing the average IBI, and then converting it to BPM and see if the value makes sense. 

For instance, say the average IBI is `700`. If you assume that these values are milliseconds (i.e., "sampling rate" of1000) then the corresponding BPM is:
`1000/700 (=1.4 beat per second) * 60 = 85.6` beats per minute, which is plausible.

marianoalbaladejo, 2021-04-01:
Finally, all my questions are resolved, thanks to everyone. The efficiency and short response time has been amazing.

closed 2021-04-01 by marianoalbaladejo