neuropsychology/NeuroKit #441: Store sampling_rate in info dict and use it in HRV

DominiqueMakowski, 2021-04-11:
The `hrv_*` functions are pretty smart and can accommodate a lot of different input formats, including a dict as returned by `ecg_/bio_process()`. I think we can start storing more information in the info dict (I also have in mind #319), such as the sampling rate. 

Then, if a dict is detected, the _hrv_sanitize_input internal can look for the `sampling_rate` key in it, and use that. This would prevent errors of people forgetting to set the sampling_rate argument, and would allow streamlined formulations like: 

```python
nk.hrv(nk.bio_process(ecg=ecg_signal, sampling_rate=200))
```

zen-juen, 2021-04-12:
For now I've added the `sampling_rate` into the info outputs of all `_process()` and `_peaks()` functions but if I recall correctly there are alot of other functions that also includes an info dict as an output (e.g., `emg_activation`) so for standardization purposes we may want them to have the same structure too. Logging here which functions have been modified: 

- `bio_process`, `ecg_process`, `rsp_process`, `emg_process`, `eda_process`, `eog_process`, `ppg_process`
- `ecg_peaks`, `eda_peaks`, `rsp_peaks`

zen-juen, 2021-04-12:
> Then, if a dict is detected, the _hrv_sanitize_input internal can look for the sampling_rate key in it, and use that. This would prevent errors of people forgetting to set the sampling_rate argument, and would allow streamlined formulations like:

If we go with this, then we might want to do it for the _analyze() functions down the road too :>

zen-juen, 2021-04-12:
Propagation of the changes in case anyone can suggest a better way of refactoring the functions 😛 

if dict or tuple is passed, `hrv_sanitize_tuple` and `hrv_sanitize_dict_or_df` respectively detect and saves the correct `sampling_rate` > in `hrv_*` specific functions, `hrv_sanitize_input` produces a tuple with peaks list + correct `sampling_rate` if the previous condition is met (otherwise, just peaks list produced) for analysis > `hrv()` automatically reflects these changes since it internally chains the domain-specific hrv functions.

```
data = nk.data("bio_resting_5min_100hz")
```

Info dict as input with and without sampling_rate specified:
```
signals, info = nk.bio_process(data, sampling_rate=100)

nk.hrv(info, sampling_rate=100)
Out[12]: 
   HRV_RMSSD  HRV_MeanNN   HRV_SDNN  ...   HRV_MSE  HRV_CMSE  HRV_RCMSE
0  38.837766  694.756381  49.036043  ...  1.589974  1.549539   1.666211

[1 rows x 55 columns]

nk.hrv(info)
Out[13]: 
   HRV_RMSSD  HRV_MeanNN   HRV_SDNN  ...   HRV_MSE  HRV_CMSE  HRV_RCMSE
0  38.837766  694.756381  49.036043  ...  1.589974  1.549539   1.666211

[1 rows x 55 columns]

```

Tuple as input with and without sampling_rate specified:

```
nk.hrv(processed, sampling_rate=100)
Out[15]: 
   HRV_RMSSD  HRV_MeanNN   HRV_SDNN  ...   HRV_MSE  HRV_CMSE  HRV_RCMSE
0  38.837766  694.756381  49.036043  ...  1.589974  1.549539   1.666211

[1 rows x 55 columns]

nk.hrv(processed)
Out[16]: 
   HRV_RMSSD  HRV_MeanNN   HRV_SDNN  ...   HRV_MSE  HRV_CMSE  HRV_RCMSE
0  38.837766  694.756381  49.036043  ...  1.589974  1.549539   1.666211

[1 rows x 55 columns]

commit ae728c26 (2021-04-12): minor fix

#441
closed 2021-04-19 by zen-juen