neuropsychology/NeuroKit #443: RSA plot

DominiqueMakowski, 2021-04-11:
All the hrv_* functions have a `show` argument that can be set to `True` to make nice plots. All but `hrv_rsa()`. We should add a plot, though I'm not sure what's the best choice. Maybe something based on cardiac coherence visualizations, or an overlay of RSP and HR? Brainstorming is welcome

Some images from google:

[image]

[image]

zen-juen, 2021-04-13:
Possible overlay of rsp and ecg signals, extracted from [here](https://www.researchgate.net/publication/333203340_Recent_development_of_respiratory_rate_measurement_technologies)

[image]

DominiqueMakowski, 2021-08-18:
@zen-juen @Tam-Pham a plot for RSA would be nice and easy to add to essentially finalize the HRV features set

DominiqueMakowski, 2021-08-18:
[image]

I'd say that a plot like that, with a) overlay of (normalized, probably a running-window normalization of like 10 seconds so that it's on the same scale throughout to help highlight the synchronization) HR & RSP rate and b) values of RSA (only when continuous RSA available) is the most efficient way of conveying the idea

closed 2022-08-15 by stale[bot]