neuropsychology/NeuroKit #444: Relevance of warning in `hrv_rsa()`

zen-juen, 2021-04-12:
There's this part of code in `hrv_rsa()` which I am abit confused about (and it might be me that wrote it, can't remember):

    if len(rsp_peaks) - len(rsp_onsets) == 0:
        rsp_peaks = rsp_peaks[:-1]
    # if len(rsp_peaks) - len(rsp_onsets) != -1:
    #     warn(
    #         "Couldn't find rsp cycles onsets and centers. Check your RSP signal.",
    #         category=NeuroKitWarning
    #     )

For now I commented out the warning, but any thoughts on whether this chunk of code is necessary?

DominiqueMakowski, 2021-04-12:
the part commented out?

closed 2022-04-09 by stale[bot]

DominiqueMakowski, 2022-04-09:
Turns out sometimes this warning was sometimes spit because of a bug that I silently fixed just yesterday in the website overhaul pr #634