neuropsychology/NeuroKit #445: Problems seeing the poincare plot

MeenaSiddharthan, 2021-04-13:
Hi~

I've been having problems viewing the poincare plot using nk! 
Here are the lines I've run:

> import matplotlib.pyplot as plt
> import numpy as np
> import neurokit2 as nk
> 
> data, sampling_rate = nk.read_bitalino("sampledata/3Dec20.txt")
> events = nk.events_find(data["LUX"], threshold=10,duration_min=60,threshold_keep='below', event_conditions=["Spontaneous", "Deep Breathing"])
> ecg_signals, ecg_info = nk.ecg_process(data["ECGBIT"], sampling_rate=sampling_rate)
> epochs = nk.epochs_create(ecg_signals, events, sampling_rate=sampling_rate, epochs_start=0, epochs_end=300)
> Spont = epochs["1"]
> peaks, info = nk.ecg_peaks(Spont, sampling_rate=sampling_rate)
> hrv_indices = nk.hrv(peaks, sampling_rate=sampling_rate, show = True)

And my output looks like below + the HRV indices are not saved as a variable:
[image]

My matplotlib version is 3.0.2
My nk version is 0.1.2
My numpy version is 1.20.1

zen-juen, 2021-04-13:
Hi @MeenaSiddharthan could you let us know if there was any error message that was printed when you ran any of the above lines? 😄

MeenaSiddharthan, 2021-04-13:
Yeah! I got this error message: 

> C:\Users\m.siddharthan\AppData\Local\Continuum\anaconda3\lib\site-packages\matplotlib\cbook\__init__.py:1377: FutureWarning: Support for multi-dimensional indexing (e.g. `obj[:, None]`) is deprecated and will be removed in a future version.  Convert to a numpy array before indexing instead.
>   x[:, None]
> C:\Users\m.siddharthan\AppData\Local\Continuum\anaconda3\lib\site-packages\matplotlib\axes\_base.py:239: FutureWarning: Support for multi-dimensional indexing (e.g. `obj[:, None]`) is deprecated and will be removed in a future version.  Convert to a numpy array before indexing instead.
>   y = y[:, np.newaxis]
> Traceback (most recent call last):
> 
>   File "<ipython-input-30-e665aa879619>", line 49, in <module>
>     hrv_indices = nk.hrv(peaks, sampling_rate=sampling_rate, show = True)
> 
>   File "C:\Users\m.siddharthan\Desktop\NeuroKit\neurokit2\hrv\hrv.py", line 108, in hrv
>     _hrv_plot(peaks, out_plot, sampling_rate, **kwargs)
> 
>   File "C:\Users\m.siddharthan\Desktop\NeuroKit\neurokit2\hrv\hrv.py", line 134, in _hrv_plot
>     ax_distrib = summary_plot(rri, ax=ax_distrib, **kwargs)
> 
>   File "C:\Users\m.siddharthan\Desktop\NeuroKit\neurokit2\stats\summary.py", line 62, in summary_plot
>     **kwargs
> 
>   File "C:\Users\m.siddharthan\AppData\Local\Continuum\anaconda3\lib\site-packages\matplotlib\__init__.py", line 1810, in inner
>     return func(ax, *args, **kwargs)
> 
> TypeError: boxplot() got an unexpected keyword argument 'manage_ticks'

zen-juen, 2021-04-13:
Seems like a matplotlib issue, could you upgrade to the latest version with `pip install matplotlib==3.4.1` and see if this works?

MeenaSiddharthan, 2021-04-13:
It works noww!! Thanks a bunch :) 
[image]

closed 2021-04-13 by MeenaSiddharthan