neuropsychology/NeuroKit #447: PPG analysis producing very large RMSSD values

nickcicero18, 2021-04-21:
**Question and context**
Hi,

I have been using NeuroKit to process PPG data collected during fMRI scans. After going through the NeuroKit PPG analysis pipeline to get rmssd values for further group-level analyses, the NeuroKit rmssd values are extremely high (most above 100, some even in the thousands). I wanted to run through the steps I am doing to ensure that no key steps are missing that may be causing these unrealistic rmssd values.

My pipeline goes as follows:

1) load PPG .txt file into pandas dataframe
2) PPG data was collected at 40Hz, so use nk.signal_interpolate() to interpolate data to 240Hz
3) bandpass filter (lowcut=0.5, highcut=8.0) using nk.ppg_clean()
4) find peaks using elgendi method within nk.ppg_findpeaks()
5) automatic artifact detection using Kubios algorithm within nk.signal_fixpeaks()
6) calculate HRV metrics using nk.hrv() on the corrected R-peaks from step 5

If you have any suggestions or see any big errors that would be very helpful.

Thanks

zen-juen, 2021-04-22:
Hi @nickcicero18 this sometimes occurs when the correct sampling rate is not properly propagated through the functions, but just to make sure, could you include a reproducible script here so that we can check through each step listed here? 😄 
Thanks!

nickcicero18, 2021-04-22:
Sure, here is the code:

`data=pd.read_table("PPG.txt"), header=None)
#interpolate from 40Hz to 240Hz
x_values=np.linspace(0, len(data)/40, num=len(data))
x_new=np.linspace(0,len(data)/40, num=len(data)*6)
data_interpolate=nk.signal_interpolate(x_values, np.array(data[0]), x_new, method='cubic')
#bandpass filter data
data_clean=nk.ppg_clean(data_interpolate, sampling_rate=240)
#find peaks
peaks = nk.ppg_findpeaks(data_clean, sampling_rate=240)
#automatic artifact detection and correction
artifacts, rpeaks_corrected =  nk.signal_fixpeaks(peaks["PPG_Peaks"], sampling_rate = 240, iterative=True,
                                                         show=False, method="Kubios")
#compute HRV metrics
hrv_indices = nk.hrv(rpeaks_corrected, sampling_rate=240, show=False)
print(hrv_indices["HRV_RMSSD"][0])`

Thanks in advance for your help!

zen-juen, 2021-04-23:
Hmmm the script looks alright to me and it doesn't seem like any of the steps is causing the unrealistically high RMSSD values to me. One thing you might like to try is perhaps using `signal_resample()` rather than `signal_interpolate()` for your preprocessing?

So rather than
```
x_values=np.linspace(0, len(data)/40, num=len(data))
x_new=np.linspace(0,len(data)/40, num=len(data)*6)
data_interpolate=nk.signal_interpolate(x_values, np.array(data[0]), x_new, method='cubic')
```
one line of code should do the trick:
```
data_interpolate=nk.signal_resample(data, sampling_rate=40, desired_sampling_rate=240, method="interpolation")
```

nickcicero18, 2021-04-25:
I tried doing nk.signal_resample() with method="interpolation" but when nk.ppg_clean() is called on the output of nk.signal_resample(), I get the following error:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/neurokit2/ppg/ppg_clean.py", line 57, in ppg_clean
    ppg_signal = as_vector(ppg_signal)
  File "/usr/local/lib/python3.6/site-packages/neurokit2/misc/type_converters.py", line 44, in as_vector
    "list of values). Current input of shape: " + str(shape)
ValueError: NeuroKit error: we expect the user to provide a vector, i.e., a one-dimensional array (such as a list of values). Current input of shape: (40500, 6)

```

It looks like the interpolation makes a nested numpy array with a shape of (40500, 6) rather than what I expected it to be which would be (40500, 1).

Any suggestions?

zen-juen, 2021-04-26:
Hey @nickcicero18 hmm the output of the `signal_resample()` should generate the same structure as the input signal. Is your raw ppg data a pandas Series / numpy array?

Here's an example of how the code should run with the ppg signal (vector with 1 column):
```
ppg = nk.data('bio_resting_5min_100hz')['PPG']  # Example data, 100 Hz
ppg_interpolated = nk.signal_resample(ppg, sampling_rate=100, desired_sampling_rate=600,
                                      method='interpolation')  # Resample to 600 Hz
ppg_clean = nk.ppg_clean(ppg_interpolated, sampling_rate=600)
ppg_peaks = nk.ppg_findpeaks(ppg_clean, sampling_rate=600)
```

nickcicero18, 2021-04-26:
I realized I had to include the column index of the pandas data frame to make it the correct shape. I no longer get any errors with your suggested workflow, but the distribution of rmssd values (even after automatic artifact detection looks like this:

[image]

This picture also cuts off two outliers with rmssd values of ~600 and ~1250. Do you have any suggestions for dealing with the rmssd values above 100? Might I have to do a manual visual inspection of each subjects time series?

Thanks

nickcicero18, 2021-04-28:
After digging some more it seems that a potential issue causing the very large RMSSD values is the ppg_findpeaks() function. When I visualized the cleaned ppg data and peaks for the file that had an RMSSD value above 1000, there were lots of missing  peaks. I compared a peak finding algorithm in a different software with the same ppg file and that one found many more peaks and also got an RMSSD value of 53. I am wondering if there are any parameters within ppg_findpeaks() or _ppg_findpeaks_elgendi() that you suggest to change/optimize to fix this?

zen-juen, 2021-04-29:
@nickcicero18  thanks for bringing this up! Let us take a look at `ppg_findpeaks()` and get back to you asap

DominiqueMakowski, 2021-04-29:
@JanCBrammer any ideas?

JanCBrammer, 2021-04-29:
@nickcicero18, running `ppg_findpeaks()` with `show=True` could give some insight into why some peaks aren't detected (it will plot the detection thresholds).

nickcicero18, 2021-04-29:
Here is an example of ppg_findpeaks() with show=True for one of the files where I am getting an extremely high rmssd value. I zoomed in to show where lots of peaks are missing. It looks like those missing peaks are below the threshold. How would I go about lowering that threshold to get ppg_findpeaks() to detect those correctly?

[image]

JanCBrammer, 2021-04-29:
> How would I go about lowering that threshold to get ppg_findpeaks() to detect those correctly?

You could directly call https://github.com/neuropsychology/NeuroKit/blob/d96f5354b60ee222561ef85f1a7b1774cb12e431/neurokit2/ppg/ppg_findpeaks.py#L69 

on the cleaned PPG and play with the `peakwindow`, `beatwindow`, `beatoffset`, `mindelay` parameters. E.g., you could try setting `beatoffset=0`.

closed 2021-12-07 by DominiqueMakowski