neuropsychology/NeuroKit #452: EDA and RSP precessing

Marina168, 2021-05-13:
I would be very grateful if you could clarify the following issues:

1.For filtering  EDA and RSP , why are you using a IIR Butterworth and not FIR ? And why SOS  method for EDA and b/a for RSP?
2.How did you decide the amplitude value 0.1 in find_peaks function and lowcut/hight=0.05 in eda_phasic function?
3.What is the difference between #Relative and # Absolute in _signal_findpeaks_keep
function and witch one is the best?
4. Why did you use cubic interpolation(and no another one) for interpolate peaks and periods?

Thanks!

zen-juen, 2021-05-13:
> 1.For filtering EDA and RSP , why are you using a IIR Butterworth and not FIR ? And why SOS method for EDA and b/a for RSP?

Alot of our signal cleaning implementations are based on established pipelines such as [Khodadad et al.](https://iopscience.iop.org/article/10.1088/1361-6579/aad7e6/meta) (RSP) and [BioSppy](https://github.com/PIA-Group/BioSPPy/blob/master/biosppy/signals/eda.py) (EDA) most of which are documented in our docstrings (Of course we're always on the look out for validated filtering methods to be incorporated into our package!)

> 2.How did you decide the amplitude value 0.1 in find_peaks function and lowcut/hight=0.05 in eda_phasic function?

Amongst the several methods in `eda_findpeaks()`, the argument of setting the minimum amplitude to 0.1 is relevant only for `neurokit` and `kim2004` method. For the `kim2004` (See [paper](http://www.etc.tuiasi.ro:81/cin/Courses/Epiom/Literarature/Kimetal.pdf) here), 0.1 represents the removal of SCRs with an amplitude smaller that 10% of the maximum SCR amplitude - this was the original authors' method and hence we followed their steps closely in our implementation. For `neurokit` method, the value of 0.1 is slightly different as it represents the removal of SCR amplitudes below 0.1 SDs from the mean of the amplitudes (also realized that the docstrings do not delineate the difference in what 0.1 means across the two methods - will update this asap). The use of 0.1 in the `neurokit` method is arguably more arbitrary as it is not based on any paper but it is similar to kim et al's in the sense that detection of peaks will depend on the level of SCR amplitude of the whole signal (rather than using an explicit threshold). Hope this clarifies!

For `eda_phasic()`, the cutoff of 0.05 was based on [BIOPAC's recommendation  ](https://www.biopac.com/knowledge-base/phasic-eda-issue/) to delineate rapid changes in skin conductance.

> 3.What is the difference between #Relative and # Absolute in _signal_findpeaks_keep
> function and witch one is the best?

For the former, the determination of peaks is based on the mean/median/max of the signal (you can specify which) whereas the latter is an explicit threshold for peak determination. You can read more from the function's [docstrings](https://neurokit2.readthedocs.io/en/latest/functions.html#neurokit2.signal.signal_findpeaks)! The choice will really depend on your objective and the nature of the signal - though I would say using relative probably allows for more objective analysis (for reasons stated in previous answer). 

> 4. Why did you use cubic interpolation(and no another one) for interpolate peaks and periods?

@DominiqueMakowski @Tam-Pham any thoughts on this? 😄

Tam-Pham, 2021-05-13:
We decided on cubic splines and not other methods (e.g. linear or quadratic) as we believe that it's the closest to the behaviour of physiological changes (smooth transition from points to points). We also had a discussion in the past to choose the best interpolation method for "biologically" plausible signal in #243

Marina168, 2021-05-13:
Thank you a lot! It is still unclear to me the choice of iir filter and not
fir, but I will keep looking. :)

DominiqueMakowski, 2021-05-13:
@Marina168 and do not hesitate to share with us the results of your research :)

Tam-Pham, 2021-05-13:
@Marina168 in general, the major reason why we pick IIR over FIR in the default processing pipelines is that IIR is just much more computationally efficient than FIR :)

Also, thank you for bringing up the choice of representation in our IIR filter. There should not be any reason why we use `ba` and `sos` for different types of signals. In fact, `sos` should be used as the default for all signals since it's supposed to have fewer problems with numerical issues [as described here](https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.filtfilt.html#scipy.signal.filtfilt). Perhaps the `ba` form was then chosen because of the compatibility reason (`sos` was only introduced later).

Marina168, 2021-05-13:
Thank you for the clarification!  :)

closed 2021-05-24 by Tam-Pham