neuropsychology/NeuroKit #459: A Question about EDA_Peak_Amplitude

RyotaTKN, 2021-06-01:
Hello,

I am planning an experiment that consists of six conditions. In a pilot study, I analyzed EDA responses by using "eda_eventrelated" function, which went well, but as for EDA_Peak_Amplitude, I got the following results.

condition1: 0.932444
condition2: 0.932444
condition3: 0.932444
condition4: 0.932444
condition5: 0.932444
condition6: 0.700357

Why is EDA_Peak_Amplitude the same among 5 conditions and only the last one different from the others?
The Onsets and Durations of these conditions were approximately the same.

Thank you in advance!

Tam-Pham, 2021-06-02:
Hi @RyotaTKN 

We would be able to help you better if you can share with us your script (that we can try to run on our side) and a sample of your data

RyotaTKN, 2021-06-02:
Thank you @Tam-Pham !

Here you can download my pliot data and here I attached.
https://www.dropbox.com/sh/q8ur52dwnguugm2/AACvG-SSOnMWhoWVijywTvLca?dl=0

Tam-Pham, 2021-06-04:
Hi @RyotaTKN 

I was able to replicate your results. Thank you so much for reporting this issue.

There is indeed a problem with the new feature in `nk.epochs_create()`. Basically, following the discussion in #413, we have added a new feature in the function to allows creation of uneven epochs from the `duration` of the events. However, this feature is newly added and we have not properly tested it. Hence, when you were using the default of `nk.epochs_create()` with `epochs_end="from_events"`, the epochs were incorrectly created, resulting in the incorrect eda results.

Thanks to your issue, we have made the correction to the function. Now, with your code, we obtain the following:

```python
import numpy as np
import pandas as pd
import neurokit2 as nk
import matplotlib.pyplot as plt
plt.rcParams['figure.figsize'] = [15, 5]

# Read data
df, sampling_rate = nk.read_acqknowledge('takanotest0531.acq')

# Find events list
fcondition_list = ["A", "A", "B", "C", "C", "B"] # There are conditions we manipulated
dfevents = nk.events_find(df["Light Sensor1"], discard_first = 0, threshold_keep="above", event_conditions=fcondition_list)

# Plot the location of event with the signals
plot = nk.events_plot(dfevents, df)

# Process data
dfeda_signals, info = nk.eda_process(df['SC - GSR100C'], sampling_rate=250)
nk.eda_plot(dfeda_signals)

dfepochs = nk.epochs_create(dfeda_signals, dfevents, sampling_rate=250)

eventtable = nk.eda_eventrelated(dfepochs)
```
```python
eventtable['EDA_Peak_Amplitude']
```
gives the following output:

1    0.291012
2    0.117552
3    0.148410
4    0.177980
5    0.728646
6    0.446083

which corresponds well to the results

[image]
[image]

Thanks again for bringing the issue to our attention. Please let us know should you run into any other issues while using the package. I will update here in this thread the changes made to the function and will let you know once the changes are updated in dev branch. You can then update the neurokit package from dev by `pip install https://github.com/neuropsychology/neurokit/zipball/dev`. I will close this issue once the changes are updated in master.

closed 2021-07-30 by Tam-Pham