neuropsychology/NeuroKit #462: HRV computation from PPG data: RMSSD values too high

tebee-hub, 2021-06-08:
**Question and context**
Hey everyone,
I am currently working on computing HRV parameters from data I obtained from a PPG sensor on the Empatica E4 wristband. Similar to #447 I am facing unrealistic results with really high RMSSD values. After doing a little research and plotting all the intermediate steps, I suspect that the problem is that with my artifacts in the input signal, the find_peaks algorithm can simply not detect enough systolic peaks in the data to produce a reliable output value for the HRV parameters. I will provide an example of my input data (a csv file with BVP data), my code to process the signal and compute HRV values and some visualization I plotted along the way to illustrate my problem. Since I know that RMSSD works with squared numbers, these big outliers that I get have a huge impact on the resulting RMSSD parameter.
One circumvention I already came up with was to use the IBI output, that Empatica computes from the BVP data themselves. There, they seem to have more consistent peaks, I don't know what their algorithm is to process the data though. I will also provide data for this approach which seems to work a little better.
So I guess my question would be if someone has a tip about what else I could try to make it work more accurately or if you can spot any mistakes I made along the way.

Thanks!

**Input data:**
[BVP_1h_64Hz.csv](https://github.com/neuropsychology/NeuroKit/files/6614693/BVP_1h_64Hz.csv)

**The code:**
import neurokit2 as nk
import pandas as pd

# Import the csv, only process second column with BVP values
set = pd.read_csv('BVP_1h_64Hz.csv')
ppg = set.iloc[:,1]
print(ppg)

# Extract the cleaned signal for peak detection
cleanPPG = nk.ppg_clean(ppg, sampling_rate=64)
print(cleanPPG)

# Find peaks
info = nk.ppg_findpeaks(cleanPPG, sampling_rate=64)
peaks = info["PPG_Peaks"]
artifacts, peaks_corrected = nk.signal_fixpeaks(peaks, sampling_rate=64, iterative=True, show=True, method="Kubios")

# Compute HRV indices
hrv_time = nk.hrv_time(peaks_corrected, sampling_rate=64)
pd.DataFrame.to_csv(hrv_time, 'HRV_time.csv')

hrv_freq = nk.hrv_frequency(peaks_corrected, sampling_rate=64)
pd.DataFrame.to_csv(hrv_freq, 'HRV_freq.csv')

**The results**
[HRV_freq.csv](https://github.com/neuropsychology/NeuroKit/files/6614702/HRV_freq.csv)
[HRV_time.csv](https://github.com/neuropsychology/NeuroKit/files/6614703/HRV_time.csv)

**BVP signal plotted:**
[image]
**Resulting distribution of R-R intervals**
[image]

**Input data IBI**
[IBI_1h_64Hz.csv](https://github.com/neuropsychology/NeuroKit/files/6614976/IBI_1h_64Hz.csv)

**Code for computing HRV on the basis of IBI values**
import neurokit2 as nk
import pandas as pd

# Import the csv, label Peaks and IBI, delete first row
ibi = pd.read_csv('IBI_1h_64Hz.csv', names=['Peaks','IBI'] , delimiter=',')
ibi = ibi.drop(ibi.index[[0]])
ibi = ibi.reset_index(drop=True)
print(ibi)

# Get peaks into a numpy array
peaks = ibi['Peaks'].to_numpy(dtype='float64')

#Automatic artifact detection and correction
artifacts, rpeaks_corrected = nk.signal_fixpeaks(peaks, sampling_rate=64, iterative=True, show=True, method="Kubios")

# Compute HRV indices
hrv_time = nk.hrv_time(rpeaks_corrected, sampling_rate=64, show=True)
print(hrv_time)
#pd.DataFrame.to_csv(hrv_time, "HRV_time_IBI.csv")

**The result**
[HRV_time_IBI.csv](https://github.com/neuropsychology/NeuroKit/files/6614987/HRV_time_IBI.csv)

Tam-Pham, 2021-06-09:
Hi @tebee-hub 

Thank you for raising the issue with us. Could you please try to set `show=True` in `nk.ppg_findpeaks`? From the plot, you can try to inspect whether there were a lot of missing systolic peaks and/or where exactly do that happen. We might have more insights into when and where it went wrong. 

Also, the `IBI_1h_64Hz.csv` file seems to be a slice of the entire signal recording or of a different recording entirely. Is the `ibi[Peaks']` supposed to be the systolic peaks computed by Empatica? It would be great if I can have the exact output for the same length of the signal to compare and see if we can improve the function to detect more accurate peaks. 

In addition, as suggested by @JanCBrammer here: https://github.com/neuropsychology/NeuroKit/issues/447#issuecomment-829502000, if the inspection shows that the function was indeed not detecting enough peaks, you can attempt to tweak the parameters directly and see if it improves the detection. Update us!

tebee-hub, 2021-06-09:
Thanks for the quick response @Tam-Pham !

I already tried setting `show=true` in the `nk.ppg_findpeaks` function. It seems that the peak detection fails in intervals with especially low amplitudes for the ppg signal wave. This would be the corresponding illustration of that:
[image]
This is also why I started tweaking the parameters, although I have to admit that my method with that was rather arbitrary. Do you have any advice on what will work there? I cannot quite grasp what each one of those four factors "peakwindow", "beatwindow", "beatoffset" and "mindelay" influences.

I am happy to supply more about the underlying data, it's true that I cut approximately 6-hour lasting recordings into slices of one hour. And you are absolutely right as well: The column ibi['Peaks'] represents systolic peaks as they are computed by Empatica on the basis of the BVP values. This would be the output file for the ~6h that I recorded.
[IBI.csv](https://github.com/neuropsychology/NeuroKit/files/6627121/IBI.csv)

I hope that this will clear things up further :)

DominiqueMakowski, 2021-06-14:
Unfortunately, no peak detection methods are perfect in noisy signal. No matter the parametrization, some portions of signal might just be too noisy to withhold any usable data, and this especially to-be-expected in very long recordings.

> [Empatica's algorithm] seem to have more consistent peaks, I don't know what their algorithm is to process the data though.

Not sure... the problem of closed-source solutions like theirs is like there's no way of telling whether it makes sense or not. Not knowing how their algorithms "finds" (or invents) these peaks makes any analyses derived from it dubious. If I were you I'd stick to transparent procedures, so that at least you know all the limits and potential issues.

Anyway, here are some ideas in regards to workarounds that can be used with overly noisy data:

- You could insert artificial peaks whenever "plausible" (see the code of `nk.signal_fixpeaks()`). This has the same limitations as data imputation; in particular, it will directly bias your HRV indices, and if the proportion of artificial peaks is too high it just won't make sense anymore.
-  Remove all the implausible IBIs (more than idk 1s, 2s or something, depending on how stringent you want to be), which is equivalent to discarding all the segments with no peaks detected. This is *relatively* fine for HRV indices that operate on shorter time-scales (rmssd), but makes little sense for those that are based on longer dynamics (which is the case of fractal or low-frequency-related indices) 
 

@JanCBrammer might have some additional thoughts

closed 2022-06-18 by stale[bot]