neuropsychology/NeuroKit #463: When calling nk.ecg_peaks() with correct_artifacts=True, the code crash in _correct_missed() function

FrancisThibaultNRC, 2021-06-11:
**Describe the bug**
- Once I call the function as the following:
_, neurokit = nk.ecg_peaks(nk.ecg_clean(ecg, sampling_rate=fs, method="neurokit"),
                                       sampling_rate=128, method="rodrigues2021",
                                       correct_artifacts=True)

the code crash in the function _correct_misssed(middsed_idcs, peaks)
at line 377
prev_peaks = corrected_peaks[[i - 1 for i in missed_idcs]]
TypeError: list indices must be integers or slices, not list

It seems that the indices created with the for is a list, which is not allowed.

**System Specifications**
install neurokit2 into a conda environment.  The python version is 3.8.6.  I run this into a pycharm IDE.
- neurokit2 library version 0.1.2

And copy paste the output:

  File "D:/git/hrv/code/scripts/1-get_physionet_raw_ecg_extract_RR_intervals.py", line 182, in RR_intervals_eval
    _, neurokit = nk.ecg_peaks(nk.ecg_clean(ecg, sampling_rate=fs, method="neurokit"),
  File "C:\ProgramData\Anaconda3\envs\test\lib\site-packages\neurokit2\ecg\ecg_peaks.py", line 74, in ecg_peaks
    _, rpeaks = signal_fixpeaks(rpeaks, sampling_rate=sampling_rate, iterative=True, method="Kubios")
  File "C:\ProgramData\Anaconda3\envs\test\lib\site-packages\neurokit2\signal\signal_fixpeaks.py", line 116, in signal_fixpeaks
    return _signal_fixpeaks_kubios(peaks, sampling_rate=sampling_rate, iterative=iterative, show=show)
  File "C:\ProgramData\Anaconda3\envs\test\lib\site-packages\neurokit2\signal\signal_fixpeaks.py", line 155, in _signal_fixpeaks_kubios
    peaks_clean = _correct_artifacts(artifacts, peaks)
  File "C:\ProgramData\Anaconda3\envs\test\lib\site-packages\neurokit2\signal\signal_fixpeaks.py", line 346, in _correct_artifacts
    peaks = _correct_missed(missed_idcs, peaks)
  File "C:\ProgramData\Anaconda3\envs\test\lib\site-packages\neurokit2\signal\signal_fixpeaks.py", line 377, in _correct_missed
    prev_peaks = corrected_peaks[[i - 1 for i in missed_idcs]]
TypeError: list indices must be integers or slices, not list

Thanks for your help.

JanCBrammer, 2021-06-13:
Hi @FrancisThibaultNRC,

Thank you for opening the issue. I could replicate the problem by running:

```python
import neurokit2 as nk

ecg = nk.ecg_simulate(duration=100, sampling_rate=128)
ecg_clean = nk.ecg_clean(ecg, sampling_rate=128, method="neurokit")
_, neurokit = nk.ecg_peaks(ecg_clean, sampling_rate=128, method="rodrigues2021",
                           correct_artifacts=True)
```

The `rodrigues2021` R-peak detector did return the R-peaks in the wrong format, causing the traceback you posted above.
The issue should now be fixed in https://github.com/neuropsychology/NeuroKit/pull/455/commits/c1f7d05c90bff6898db1271d3d8ac3e8002ad67f.

FrancisThibaultNRC, 2021-06-13:
Thank you much for your quick reply.

closed 2021-12-11 by DominiqueMakowski