neuropsychology/NeuroKit #465: bug in _ecg_findpeaks_rodrigues function

FrancisThibaultNRC, 2021-06-14:
**Describe the bug**
out of bounds happens in the function __ecg_findpeaks_rodrigues while doing something simple.

**To Reproduce**

    duration = 600; sampling_rate = 500; heart_rate = 50; noise = 0.0; method='simple'
    ecg = nk.ecg_simulate(duration=duration,
                          sampling_rate=sampling_rate,
                          noise=noise,
                          heart_rate=heart_rate,
                          method=method,
                          random_state=10)
    ecg_clean = nk.ecg_clean(ecg, sampling_rate=sampling_rate)
    _, neurokit = nk.ecg_peaks(ecg_clean, sampling_rate=sampling_rate, method="rodrigues2021")

**System Specifications**
python 3.8.6
neurokit2 version 0.1.2
OS windows 10

And copy paste the output:
    def test_test():
        """Test if an ecg simulated signal of 50 and 100 bpm will result in a mean heart rate feature extraction
         of the same values"""

        plot_simulated_data = False

        # Simulate data: heart beat = 50 and 100 bpm
        duration = 600; sampling_rate = 500; heart_rate = 50; noise = 0.0; method='simple'
        ecg = nk.ecg_simulate(duration=duration,
                              sampling_rate=sampling_rate,
                              noise=noise,
                              heart_rate=heart_rate,
                              method=method,
                              random_state=10)
        ecg_clean = nk.ecg_clean(ecg, sampling_rate=sampling_rate)
>       _, neurokit = nk.ecg_peaks(ecg_clean, sampling_rate=sampling_rate, method="rodrigues2021")

code\scripts\plugin\test_hrv_plugin.py:1041:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
C:\ProgramData\Anaconda3\envs\hop-child-covid\lib\site-packages\neurokit2\ecg\ecg_peaks.py:71: in ecg_peaks
    rpeaks = ecg_findpeaks(ecg_cleaned, sampling_rate=sampling_rate, method=method)
C:\ProgramData\Anaconda3\envs\hop-child-covid\lib\site-packages\neurokit2\ecg\ecg_findpeaks.py:157: in ecg_findpeaks
    rpeaks = _ecg_findpeaks_rodrigues(ecg_cleaned, sampling_rate)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

signal = array([0.26803822, 0.29552381, 0.32460534, ..., 0.04503541, 0.04322261,
       0.04140278]), sampling_rate = 500

    def _ecg_findpeaks_rodrigues(signal, sampling_rate=1000):
        """Segmenter by Tiago Rodrigues, inspired by on Gutierrez-Rivas (2015) and Sadhukhan (2012).

        References
        ----------
        - Gutiérrez-Rivas, R., García, J. J., Marnane, W. P., & Hernández, A. (2015). Novel real-time
          low-complexity QRS complex detector based on adaptive thresholding. IEEE Sensors Journal,
          15(10), 6036-6043.

        - Sadhukhan, D., & Mitra, M. (2012). R-peak detection algorithm for ECG using double difference
          and RR interval processing. Procedia Technology, 4, 873-877.

        """

        N = int(np.round(3 * sampling_rate / 128))
        Nd = N - 1
        Pth = (0.7 * sampling_rate) / 128 + 2.7
        # Pth = 3, optimal for fs = 250 Hz
        Rmin = 0.26

        rpeaks = []
        i = 1
        tf = len(signal)
        Ramptotal = 0

        # Double derivative squared
        diff_ecg = [signal[i] - signal[i - Nd] for i in range(Nd, len(signal))]
        ddiff_ecg = [diff_ecg[i] - diff_ecg[i - 1] for i in range(1, len(diff_ecg))]
        squar = np.square(ddiff_ecg)

        # Integrate moving window
        b = np.array(np.ones(N))
        a = [1]
        processed_ecg = scipy.signal.lfilter(b, a, squar)

        # R-peak finder FSM
        while i < tf - sampling_rate:  # ignore last second of recording

            # State 1: looking for maximum
            tf1 = np.round(i + Rmin * sampling_rate)
            Rpeakamp = 0
            while i < tf1:
                # Rpeak amplitude and position
                if processed_ecg[i] > Rpeakamp:
                    Rpeakamp = processed_ecg[i]
                    rpeakpos = i + 1
                i += 1

            Ramptotal = (19 / 20) * Ramptotal + (1 / 20) * Rpeakamp
            rpeaks.append(rpeakpos)

            # State 2: waiting state
            d = tf1 - rpeakpos
            tf2 = i + np.round(0.2 * 2 - d)
            while i <= tf2:
                i += 1

            # State 3: decreasing threshold
            Thr = Ramptotal
>           while processed_ecg[i] < Thr:
E           IndexError: index 299988 is out of bounds for axis 0 with size 299988

Thanks you for your collaboration.

Best.

JanCBrammer, 2021-06-14:
@TiagoTostas, would you mind taking a look at this?

TiagoTostas, 2021-07-11:
@FrancisThibaultNRC thank you for pointing out this issue.

It appears that in cases where the 3rd Stage of the FSM is longer than expected (in particular for low HR),  there can be situations where the index of the signal is out of range during the exponential decay update phase.

I will make a PR to address the issue.

Best,

FrancisThibaultNRC, 2021-09-28:
Hi all,

I am still having the same issue using different parameters.  See the test with different parameters:

closed 2021-09-28 by FrancisThibaultNRC