neuropsychology/NeuroKit #466: Zhao2018 quality estimate function has wrong window length defaults

Miceuz, 2021-06-17:
**Describe the bug**
In documentation it is written

> The "zhao2018" method (Zhao et la., 2018) was originally designed for signal with a length of 10 seconds.

However default value for window is 1024:
`def _ecg_quality_zhao2018(ecg_cleaned, rpeaks=None, sampling_rate=1000,
                          window=1024, kurtosis_method="fisher", mode="simple", **kwargs)`

While stating that `window` argument is in seconds, not milliseconds.

>window : int
>        Length of each window in seconds. See `signal_psd()`.

Also `signal_psd` stated that window is in seconds:
>window : int
>        Length of each window in seconds (for Welch method). If None (default), window will be automatically
>        calculated to capture at least 2 cycles of min_frequency. If the length of recording does not
>        allow the formal, window will be default to half of the length of recording.

zen-juen, 2021-06-18:
Hi there @Miceuz thanks for raising this issue! Let me take a look at the original [paper](https://www.frontiersin.org/articles/10.3389/fphys.2018.00727/full) that implemented this algorithm and get back to you 😄

zen-juen, 2021-06-18:
Hmm looking into the database that the original authors used (they used two databases from the Physionet/Cinc Challenge 2011 and [Physionet/Cinc Challenge 2017](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC5978770/)), it doesn't seem like the signals used were 10 seconds in length. For example, the 2017 database contains signals obtained from "a single short ECG lead recording (between 30 s and 60 s in length)" so I'm not too sure why we documented that the algorithm was designed for specifically a 10s-long signal 🤔 I might be missing something here but it seems to me that this algorithm doesn't have to be used for only 10s signals (if @DominiqueMakowski can confirm 😄 we can then correct this documentation)

Regarding the `window` argument which defaults to 1024 (which relates to the segments of samples used) this is an internal function specifically to calculate the power spectrum density (used in `nk.signal_psd()`) which is then passed to calculate the quality of the signal - so this has nothing to do with the length of the ECG signal. Hope this clarifies!

DominiqueMakowski, 2021-06-18:
I think we took the description from [biosppy's](https://github.com/PIA-Group/BioSPPy/blob/master/biosppy/signals/ecg.py#L1679) without double checking it, but yeah if it's not correct then we need to change.

Miceuz, 2021-08-02:
My original idea when posting this bug report was that _ecg_quality_zhao2018 assumes window is 1024 while from user-facing API it's not clear what is the minimum length the recording has to be. In fact at the bottom `signal_psd` requires the signal to be at least 1024, call would fail with smaller window sizes.

Also, judging from `signal_psd` documentation `window` is in seconds, not in samples.

closed 2022-09-20 by stale[bot]