neuropsychology/NeuroKit #467: ECG simulation method based on deepfake-ecg

DominiqueMakowski, 2021-06-18:
Can add this as a method for ECG simulation: https://github.com/vlbthambawita/deepfake-ecg

We'd need to check for the installation of that package (and if not tell to run `pip install deepfake-ecg`) and then run

```python
import deepfakeecg

#deepfakeecg.generate("number of ECG to generate", "Path to generate", "start file ids from this number", "device to run") 

deepfakeecg.generate(5, ".", start_id=0, run_device="cpu") # Generate 5 ECGs to the current folder starting from id=0
```

If no possibility to directly load the data (i.e., if it can only save a file on the disk), we can save the file, load it and then delete it.

zen-juen, 2021-06-18:
We can probably do something like this
```
def _ecg_simulate_deepfake(n_heartbeats=5):

    # Try loading deepfakeecg
    try:
        import deepfakeecg
    except ImportError:
        raise ImportError(
            "NeuroKit error: _ecg_simulate_deepfake(): the 'deepfakeecg' module is required ",
            "for this function to run. ",
            "Please install it first (`pip install deepfakeecg`).",
        )

    # Generate n_heartbeats ECGs to the current folder starting from id=0
    deepfakeecg.generate(n_heartbeats, ".", start_id=0, run_device="cpu") 

    current_path = os.getcwd()

    data_list = []
    for i in range(n_heartbeats):
        file_name = str(i) + '.asc'
        file = os.path.join(current_path, file_name)
        data = np.loadtxt(file, skiprows=0)
        ecg = np.concatenate(data)
        data_list.append(ecg)

        # remove files
        os.remove(file)

    return data_list
        
```
with data_list being something like
Out[13]: 
[array([ -61.,  169.,  -62., ...,  -11., -113.,  -37.]),
 array([-162.,  113.,   -9., ...,  -15.,  -86.,  -35.]),
 array([ -63.,  -32.,  -24., ...,  -74., -150., -125.]),
 array([-144.,   19.,   30., ...,  175.,  181.,  164.]),
 array([ -39., -137.,   79., ...,   -5.,  -77.,  -63.])]
```

Based on the `deepfakeecg` [source code](https://github.com/vlbthambawita/deepfake-ecg/blob/main/deepfakeecg/deepfakeecg.py), the only variable parameter is the number of signals (specified as `n_heartbeats` here) that you can generate. I'm not sure what's the best way of incorporating it into our ecg module so that it is consistent with the rest of the other internal methods. In this example here I store 5 ecg signals (one array being one signal) in `data_list` - from this, we can either randomly choose one signal to return as the final simulated output or we can also scrap the `n_heartbeats` argument entirely and just default to returning one signal.

DominiqueMakowski, 2021-06-18:
so each "signal" is one QRS complex? In that case I suppose we can assemble them to make a continuous signal with multiple beats?

zen-juen, 2021-06-19:
Nope one signal is the full signal itself

```
pd.Series(data_list[0]).plot()
```

[image]

alexxony, 2021-08-13:
There are some missing values Due to bluetooth disconnection.

Can this make complement for missing value area?

I thinks it is beneficial for R peaks detection

zen-juen, 2021-08-13:
Hi @alexxony I don't get quite what you mean here, do you mean with this specific implementation of simulating ecg using `deepfake`?

closed 2022-08-10 by stale[bot]