neuropsychology/NeuroKit #472: Meet an IndexError when using method ecg_peaks

skywalkerzhang, 2021-06-26:
**To Reproduce**
I encounter the IndexError when use this(shown below) to get the QRS interval
```python
_, r_peaks = nk.ecg_peaks(ecg_signal, sampling_rate=200)
```
The error stack is shown as below:
```
Traceback (most recent call last):
  File "C:\Users\16778\anaconda3\lib\site-packages\IPython\core\interactiveshell.py", line 3437, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-2-5f81baaf3b97>", line 1, in <module>
    runfile('C:/Users/16778/Documents/ECG/python-classifier-2021/QRS_interval.py', wdir='C:/Users/16778/Documents/ECG/python-classifier-2021')
  File "C:\Users\16778\scoop\apps\pycharm-professional\current\IDE\plugins\python\helpers\pydev\_pydev_bundle\pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "C:\Users\16778\scoop\apps\pycharm-professional\current\IDE\plugins\python\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "C:/Users/16778/Documents/ECG/python-classifier-2021/QRS_interval.py", line 24, in <module>
    _, r_peaks = nk.ecg_peaks(ecg_signal, sampling_rate=200)
  File "C:\Users\16778\anaconda3\lib\site-packages\neurokit2\ecg\ecg_peaks.py", line 71, in ecg_peaks
    rpeaks = ecg_findpeaks(ecg_cleaned, sampling_rate=sampling_rate, method=method)
  File "C:\Users\16778\anaconda3\lib\site-packages\neurokit2\ecg\ecg_findpeaks.py", line 135, in ecg_findpeaks
    rpeaks = _ecg_findpeaks_neurokit(ecg_cleaned, sampling_rate, show=show)
  File "C:\Users\16778\anaconda3\lib\site-packages\neurokit2\ecg\ecg_findpeaks.py", line 260, in _ecg_findpeaks_neurokit
    end_qrs = end_qrs[end_qrs > beg_qrs[0]]
IndexError: index 0 is out of bounds for axis 0 with size 0
```

Thank you in advance for helping me.

zen-juen, 2021-06-27:
Seems like the error is occurring during the detection of the beginning of each QRS, which means you probably have an empty `beg_qrs` (cannot index an empty array). 

Is the passed `ecg_signal` the raw signal recorded or the cleaned signal (with `ecg_clean()`)? 
Could you also print `ecg_signal.shape` and `ecg_signal.dtype` here so that we can see if it's the nature of the input that's causing this?

skywalkerzhang, 2021-06-28:
yes -- I use ECG clean
the shape is (5000,) and the type is float64
do you need the raw data? I save it as npz.

zen-juen, 2021-06-28:
Yes, it'll be great if you can share with us the data and a reproducible script here so that we can investigate 😄

skywalkerzhang, 2021-06-28:
[error_data.zip](https://github.com/neuropsychology/NeuroKit/files/6722800/error_data.zip)
This is the data
```python
import neurokit2 as nk
import numpy as np

ecg_signal = np.load('../data/error_data.npy')
ecg_signal = nk.ecg_clean(ecg_signal, 500)
_, r_peaks = nk.ecg_peaks(ecg_signal, sampling_rate=500)
```

My raw data is very large so I use try except to get the error data
If I provide the wrong data, which I mean the data cannot generate index error. Please connect me.
Thank you so much

zen-juen, 2021-06-28:
The raw data gives an array filled with zeros - could this be a recording issue?

```
ecg_signal
Out[225]: array([0., 0., 0., ..., 0., 0., 0.])
```

DominiqueMakowski, 2021-06-28:
are there only zeros? `len(np.unique(ecg_signal))`

skywalkerzhang, 2021-06-28:
> are there only zeros? `len(np.unique(ecg_signal))`

You are right. It is my fault. Deeply sorry.

skywalkerzhang, 2021-06-28:
> The raw data gives an array filled with zeros - could this be a recording issue?
> 
> ```
> ecg_signal
> Out[225]: array([0., 0., 0., ..., 0., 0., 0.])
> ```

Really thank you for your help. Sorry for the trouble I cause.

DominiqueMakowski, 2021-06-28:
no worries, cheers!

DominiqueMakowski, 2021-06-28:
@zen-juen that said we could also throw an informative error in case there are no peaks, like "no peaks were detected, check your signal" or something

closed 2021-06-28 by DominiqueMakowski