neuropsychology/NeuroKit #476: Could you please tell me where to find the principle of neurokit" in ecg_peaks_method.

skywalkerzhang, 2021-07-07:
**Question and context**
I'm working on my graduation project, and I want to introduce the principle of peak detection of your project in the method section, but I haven't found it in the documents and cited papers you provided (which will be quoted in the graduation project). May I ask this principle? Can it be made public?

DominiqueMakowski, 2021-07-07:
You mean the default peak detection method for... ECG?

skywalkerzhang, 2021-07-07:
> You mean the default peak detection method for... ECG?

Yes. I couldn't find the paper.

DominiqueMakowski, 2021-07-07:
You can find more details in this discussion here https://github.com/neuropsychology/NeuroKit/issues/245

The current default method has been written by @JanCBrammer but afaik there is not yet a citation available that validates this algorithm in particular (though for now you can cite [biopeaks](https://github.com/JanCBrammer/biopeaks) and neurokit)

JanCBrammer, 2021-07-07:
See JanCBrammer/biopeaks#23.

closed 2021-07-28 by DominiqueMakowski