neuropsychology/NeuroKit #477: Why my ECG_R_Peaks is empty

skywalkerzhang, 2021-07-07:
**Question and context**
Hi, I've got empty ECG_R_Peaks by invoking nk.ecg_peaks. I wonder why this happens.

Here is a sample code and sample input:

```python
import neurokit2 as nk
import numpy as np
ecg_signal = np.loadtxt('./test.txt')
_, r_peaks = nk.ecg_peaks(ecg_signal, sampling_rate=257)
print(r_peaks)
```
[test.txt](https://github.com/neuropsychology/NeuroKit/files/6778248/test.txt)

zen-juen, 2021-07-09:
Hmm I suspect that this may be due to the incredibly high heart rate detected from this signal. Based on a sampling rate of 257, this graph shows a signal of duration 10s and if I were to arbitrarily count the number of peaks as those exceeding 100 (`len(np.where(ecg_signal > 100)[0])`), this gives 271 peaks in 10s - which is unrealistic 😅 

Can you double check if the sampling rate is indeed 275?

[image]

DominiqueMakowski, 2021-07-09:
Unless it's a [hummingbird](https://dinoanimals.com/animals/heart-rates-of-animals-top-10/) which heart rate is ~1260 bpm 🤷

skywalkerzhang, 2021-07-10:
> Hmm I suspect that this may be due to the incredibly high heart rate detected from this signal. Based on a sampling rate of 257, this graph shows a signal of duration 10s and if I were to [...]

Sorry for my mistake. The dataset that I use has some noise data. -- I guess it's one of the noise data.

I try to ignore it and just give a prediction of this kind of data. But I think I should deal with these data now.

closed 2021-07-17 by zen-juen