neuropsychology/NeuroKit #479: nk.ecg_quality loop

alexxony, 2021-07-13:
**Question and context**

I have 48 hours ECG data; there are many motion artifacts. 
I planned nk.ecg_quality loop, long term calculation  can not detect motion artifact. So I tried to make sliding window approach. 
window size is 7 * frequency. 

1.  len_ecg=len(ecg_cleaned) 
    step_half=int(f/2) # f is 256
    f_7=int(7*f)

2.  x=deque()
    for i in islice(range(len_ecg), 0, len_ecg, step_half):
         x.append(i)
3. idx=[]
   y=[]

   while x:
        if nk.ecg_quality(ecg_cleaned[x.popleft():f_7+x.popleft()], sampling_rate=f, method="zhao2018", approach="fuzzy")=='Excellent':
        y.append(1)
        idx.append(x.popleft())
       else:
        y.append(0)
        idx.append(x.popleft())

when I runned at 1st, 

IndexError: index 0 is out of bounds for axis 0 with size 0
and len(y)=8814

so I set range [x[8814]:f_7+x[8814]] 

and run nk.ecg_quality(ecg_cleaned[x[8814]:f_7+x[8814]] , sampling_rate=f, method="zhao2018", approach="fuzzy")
its result was 'Excellent' 
[x[8815]:f_7+x[8815]] was also

it means IndexError was not real...

I reset the whole ipynb and run the code again.

IndexError: index 0 is out of bounds for axis 0 with size 0
and len(y)=5410

I don't know why this error happen

sotkal, 2021-07-22:
I have been working with this function with an other approach, but got the same error when the lower level function **_ecg_quality_zhao2018**  used in **ecg_quality**, couldnt detect any R peaks in a time-window. If I remember right **_ecg_quality_zhao2018** uses the default neurokit peak-detection method without artifact correction. Run the default peakfinder in those segments and see if any peaks are detected.
Hope this helps.

alexxony, 2021-07-23:
> I have been working with this function with an other approach, but got the same error when the lower level function **_ecg_quality_zhao2018** used in **ecg_quality**, couldnt detect any R peaks in a [...]

can you upload the example coding???

bhralzz, 2022-05-02:
without any rep!????????!

closed 2022-12-03 by stale[bot]