neuropsychology/NeuroKit #480: nk.ecg_delineate do not show Q and S peaks.

HelenaRoMar, 2021-07-14:
Hello!!

I was experimenting with the characteristics of Neurokit2 and doing the examples I have some problem with the ecg_delineate feature.
I used:

# Delineate the ECG signal
_, waves_peak = nk.ecg_delineate(ecg_signal, rpeaks, sampling_rate=fs)

I followed this[ example](https://neurokit2.readthedocs.io/en/latest/examples/ecg_delineate.html)

So I don't understand why only T and P peaks appear.

Thank you in advande for your help.

zen-juen, 2021-07-17:
Hi @HelenaRoMar if you didn't use one of neurokit's example dataset for this, can you share with us a sample of your data and a reproducible script so that we can investigate the error?

feliperbroering, 2021-07-19:
Hello, I observed the same problem, I tested with my samples and also with the neurokit dataset.

The S and Q peaks are always missing.

[image]

Tam-Pham, 2021-07-19:
Hi  @HelenaRoMar and @feliperbroering 

Thank you for checking in with us. Please note that this behaviour is expected at the current stage of the function. As described [here](https://neurokit2.readthedocs.io/en/latest/functions.html#neurokit2.ecg.ecg_delineate), the current default method in `ecg_delineate()` is `discrete wavelet method (dwt)` and it only returns the following: P-peaks, T-peaks, P-onsets, P-offsets, T-onsets, T-offsets, QRS-onsets and QRS-offsets .

The only one method that support the return of Q-peak and S-peak is the derivative method (`method="peak`). We do hope to further improve the wavelet methods to be able to return these peaks information soon. But for now, for these 2 peaks, please run the delineate function but change the method to `peak` instead of `dwt`.

HelenaRoMar, 2021-07-19:
Hi @Tam-Pham 

Thank you so much for your response. 

It was a confusion when I read in the documentation:  _‘peak’ (default)_ 
[image]

But using:  _method="peak"_
```
ecg = nk.ecg_simulate(duration=10, sampling_rate=1000)
cleaned = nk.ecg_clean(ecg, sampling_rate=1000)
_, rpeaks = nk.ecg_peaks(cleaned, sampling_rate=1000)
signals, waves = nk.ecg_delineate(cleaned, rpeaks, sampling_rate=1000, method="peak", show=True)
```

It works perfectly:
[image]

feliperbroering, 2021-07-19:
Hello @Tam-Pham and @HelenaRoMar 

Thanks for your feedback! 

I think is just a doc issue, because the 0.1.0 version works exactly as described in the example:
https://neurokit2.readthedocs.io/en/latest/examples/ecg_delineate.html

With the 0.1.2 version you must to explicitly use `method="peak` to works.

In 0.1.2 version:
[image]

In the 0.1.0 version:
[image]

In the 0.1.2 version forcing "method="peak":
[image]

Tam-Pham, 2021-07-22:
We indeed have only recently changed the default method in `ecg_delineate()` from "peak" to "dwt". It's strange that the documentation is not yet updated. I will check again. Thanks for letting us know! 

I will close this issue as of now :)

closed 2021-07-22 by Tam-Pham

chepo92, 2021-07-25:
Same issue with the default notebook example, I get ` KeyError: 'ECG_Q_Peaks' ` thanks for the clarification on the method to use. Also I think the readthedocs automation is may not be working https://github.com/neuropsychology/NeuroKit/blob/master/docs/examples/ecg_delineate.ipynb