neuropsychology/NeuroKit #481: `ecg_delineate` fails with KeyError if rpeak is too close to the start

timvlaer, 2021-07-16:
**Describe the bug**
If you execute the ecg_delineate function on a dataset with an rpeak close to the start, it fails with a KeyError.

**To Reproduce**

```python
ecg = pd.read_csv('period_ecg.csv') # see attached dataset below
rp, rpeaks = nk.ecg_peaks(ecg['BP'], sampling_rate=250)
nk.ecg_delineate(ecg['BP'], rpeaks, sampling_rate=250, show=False, method="peak")
```
fails with
```
KeyError: '[-20] not in index'
```

**Expected behaviour**
No exception, but the regular output of the delineate function. If the p-wave is not in the ecg signal, it should not be in the result set. 

**System Specifications**

It's important that you give us some information about the system you are using. For that you can run:

```python
>>> nk.version()
- OS: Darwin ( 64bit)
- Python: 3.9.5
- NeuroKit2: 0.1.2

- NumPy: 1.19.5
- Pandas: 1.2.3
- SciPy: 1.6.1
- sklearn: 0.24.2
- matplotlib: 3.4.2
```

Example dataset:
[period_ecg.csv](https://github.com/neuropsychology/NeuroKit/files/6829038/period_ecg.csv)

timvlaer, 2021-07-16:
Should anyone hit this issue, the workaround I follow right now is to filter out the first rpeak if it's below a certain threshold (index depends on your sample rate) 

```python
filtered_peaks = list(filter(lambda r: r > 150, rpeaks["ECG_R_Peaks"]))
nk.ecg_delineate(ecg, filtered_peaks, sampling_rate=250, show=False, method="peak") 
```

zen-juen, 2021-07-17:
Ah yes this is because the signal started after the P onset of the first heartbeat. Did a quick inspection on the first 1000 samples of the signal you provided and you can see here that the first value of `ECG_P_Onsets` is negative.

```
waves
Out[182]: 
{'ECG_P_Peaks': [39, 330, 614, 895],
 'ECG_Q_Peaks': [71, 360, 645, 932],
 'ECG_S_Peaks': [92, 380, 667, 951],
 'ECG_T_Peaks': [111, 398, 682, 967],
 'ECG_P_Onsets': [-21, 316, 600, 889],
 'ECG_T_Offsets': [136, 426, 696, 990]}
```
@DominiqueMakowski  @Tam-Pham since `signal_formatpeaks` can only run on positive indices, we can make some minor code adjustments in `ecg_delineator` by forcing negative indices to be dropped (line 131)
```
waves_noNA[feature] = [int(x) for x in waves_noNA[feature] if ~np.isnan(x) and x > 0]
```
...though I feel like this issue has been brought up before? 😅 can't remember we already have another sureproof way of dealing with this since this is likely to be a common phenomenon

closed 2021-08-24 by Tam-Pham