neuropsychology/NeuroKit #488: Benchmark of ecg_clean

alexxony, 2021-08-05:
**Question and context**

I would like to benchmark of ecg_clean

"ECG_NeuroKit"/ "ECG_BioSPPy"/ "ECG_PanTompkins"/ "ECG_Hamilton"/ "ECG_Elgendi"/ "ECG_EngZeeMod"/etc

etc: neurokit have another module?

but i can't find any example of benchmark

DominiqueMakowski, 2021-08-05:
See this discussion: https://github.com/neuropsychology/NeuroKit/issues/245

Let us know if you obtain any interesting findings

alexxony, 2021-08-06:
> See this discussion (pertaining to the peak detection, rather than on the cleaning): #245
> 
> We don't have benchmarks for cleaning, so let us know if you obtain any interesting findings!

I read the benchmark of R-peak detection which is a kind of feature extraction. however, is there any evaluation of cleaning methods?  For example,  Rini Thakur et al, wrote a benchmark paper of ecg cleaning.
Review of noise removal techniques in ECG signals
DOI: 10.1049/iet-spr.2020.0104

They used  root-mean-square error (RMSE), percentage-root-mean-square difference (PRD), and improvement in signal-to-noise ratio (SNRimp) as Performance parameters. 

Can I find any example of ecg cleaning benchmark?

DominiqueMakowski, 2021-08-06:
We don't have any benchmark data for the currently available ECG cleaning methods yet, so I'm afraid you'll have to produce it yourself

closed 2021-08-16 by DominiqueMakowski