neuropsychology/NeuroKit #492: resample for looping ecg_quality function

alexxony, 2021-08-13:
**Question and context**

I am applying 

**nk.ecg_quality(x, sampling_rate=1000, method="zhao2018", approach="fuzzy")**

on my pandas data.

my data is originally 256 Hz data; I will call my data "x"

I made a function for detect bad quality ecg area.

**def zhao (x):
    if nk.ecg_quality(x, sampling_rate=1000, method="zhao2018", approach="fuzzy")=='Excellent':
        return 1
    else:
        return 0**

And I applied this on

**pdf['rolling_col'] = pdf.loc[:, ['ECG_Raw']].rolling(5000).apply(zhao).shift(-5000)**

but thousands epochs went on, there was a index error

_IndexError: index 0 is out of bounds for axis 0 with size 0_

 So I used 

**upsampled_fft = nk.signal_resample(x, method="FFT",sampling_rate=256, desired_sampling_rate=1000)**

So I tried again, it works.

But I am afraid of distortion in analysis. 

Do you have any solution without resample??

PS. I tried to dismantle neurokit's functions but it was too hard to me

+ Have you ever collaborate in pyspark?

zen-juen, 2021-08-13:
Hi @alexxony the index error often occurs when the `sampling_rate` is not properly propagated in the functions. Your current `zhao` function uses a sampling rate of 1000, so you'll need to tweak this so that `sampling_rate` is passed in zhao, like so:

```
def zhao (x, sampling_rate):
    quality = nk.ecg_quality(x, sampling_rate=sampling_rate, method="zhao2018", approach="fuzzy")
    if quality == 'Excellent':
        return 1
    else:
        return 0
```
and then calling `zhao(x, sampling_rate=256)` should work. 

And nope we have never collaborated with pyspark!

alexxony, 2021-08-17:
> Hi @alexxony the index error often occurs when the `sampling_rate` is not properly propagated in the functions. Your current `zhao` function uses a sampling rate of 1000, so you'll need to tweak this so that `sampling_rate` is passed, like so:
> 
> ```
> def zhao (x, sampling_rate):
>     quality = nk.ecg_quality(x, sampling_rate=sampling_rate, method="zhao2018", approach="fuzzy")
>     if quality == 'Excellent':
>         return 1
>     else:
>         return 0
> ```
> 
> and then calling `zhao(x, sampling_rate=256)` should work.
> 
> And nope we have never collaborated with pyspark!

Thanks for replying.

I found the reason of error.
It was not frequency but interval and data size.
interval is too small (under 1000)/ data size is too big (over millions rows)
There is index errors.

here is my corrected code

v_test=v[0:100000]

def zhao (x):
    if nk.ecg_quality(x, sampling_rate=f, method="zhao2018", approach="fuzzy")=='Excellent':
        return 1
    else:
        return 0

pdf = pd.DataFrame(v_test)

pdf.columns =['ECG_Raw']

start = time.time()  
pdf['rolling_col'] = pdf.loc[:, ['ECG_Raw']].rolling(1792).apply(zhao).shift(-1792)
print("time :", time.time() - start)  

a=pdf['ECG_Raw'].to_numpy()*100
b=pdf['rolling_col'].to_numpy()

plt.figure(figsize=(36,9))
plt.plot(a)
plt.plot(b)
plt.show()

[image]

plt.figure(figsize=(36,9))
plt.plot(a)
plt.plot(b)
plt.xlim(6000,8000)
plt.show()

[image]

I shall apply this on spark udf

and interval size; is there any standard which is referred on papers?

zen-juen, 2021-08-18:
I'm not sure if there is any gold standard related to this but you can take a look at the original [Zhao 2018](https://www.frontiersin.org/articles/10.3389/fphys.2018.00727/full) paper or papers that have cited it to see if you can get some ideas!

closed 2021-09-04 by zen-juen