neuropsychology/NeuroKit #496: graph plotting now working any more

shashanka300, 2021-08-19:
signal_plot function dosent plot any more

zen-juen, 2021-08-19:
Hi @shashanka300 could you please elaborate on this and give us a reproducible example so that we can investigate?

shashanka300, 2021-08-19:
> Hi @shashanka300 could you please elaborate on this and give us a reproducible example so that we can investigate?

shashanka300, 2021-08-19:
sorry closed by mistake, kindly run the example, apparently the plot function has stoped working

zen-juen, 2021-08-19:
No worries! What's your neurokit version? Can you also show us what is the error message you encountered?

shashanka300, 2021-08-19:
> No worries! What's your neurokit version? Can you also show us what is the error message you encountered?

[image]

normally post execution i used to get graphs, now nothing comes no error either

zen-juen, 2021-08-20:
Are you using jupyter notebook as your IDE? You might need a `%matplotlib inline` at the start to see plots in your interface (see [more options](https://medium.com/@Med1um1/using-matplotlib-in-jupyter-notebooks-comparing-methods-and-some-tips-python-c38e85b40ba1))

Manthan-tech, 2022-06-07:
Is there any solution for this, I am facing the same issue with the latest version. Nothing is happening on running signal_plot, ecg_plot, event_plot function

shashanka300, 2022-06-07:
simple solution use plt.show() EOL

Manthan-tech, 2022-06-08:
> simple solution use plt.show() EOL

My bad.., Thanks it worked😅

closed 2022-09-22 by stale[bot]