neuropsychology/NeuroKit #498: Running _ecg_quality_pSQI, _ecg_quality_basSQI

alexxony, 2021-08-19:
**Question and context**
I tried to dismantle neurokit2.ecg.ecg_quality for speed up
but I can not run
_ecg_quality_pSQI, _ecg_quality_basSQI

pdf is ecg_3000hz.csv

@nb.jit 
def _ecg_quality_pSQI(
    ecg_cleaned,
    sampling_rate=1000,
    window=1024,
    num_spectrum=[5, 15],
    dem_spectrum=[5, 40],
    **kwargs
):
    """Power Spectrum Distribution of QRS Wave."""

    psd = signal_power(
        ecg_cleaned,
        sampling_rate=sampling_rate,
        frequency_band=[num_spectrum, dem_spectrum],
        method="welch",
        normalize=False,
        window=window,
        **kwargs
    )

    num_power = psd.iloc[0][0]
    dem_power = psd.iloc[0][1]

    return num_power / dem_power

@nb.jit 
def _ecg_quality_basSQI(
    ecg_cleaned,
    sampling_rate=1000,
    window=1024,
    num_spectrum=[0, 1],
    dem_spectrum=[0, 40],
    **kwargs
):
    """Relative Power in the Baseline."""
    psd = signal_power(
        ecg_cleaned,
        sampling_rate=sampling_rate,
        frequency_band=[num_spectrum, dem_spectrum],
        method="welch",
        normalize=False,
        window=window,
        **kwargs
    )

    num_power = psd.iloc[0][0]
    dem_power = psd.iloc[0][1]

    return 1 - num_power / dem_power

_ecg_quality_pSQI(pdf,sampling_rate=1000, window=1024, num_spectrum=[5, 15], dem_spectrum=[5, 40] )

it raised error

**UnsupportedError: Failed in object mode pipeline (step: analyzing bytecode)
Use of unsupported opcode (BUILD_MAP_UNPACK_WITH_CALL) found

File "<ipython-input-30-60bbb555048b>", line 21:
def _ecg_quality_pSQI(
    <source elided>

    psd = signal_power(**

how can i use these functions?

DominiqueMakowski, 2021-08-19:
On a side note, please try to edit your post using the markdown syntax so that the different code chunks are clearly specified, otherwise it makes it fairly difficult to read

alexxony, 2021-08-20:
def _ecg_quality_pSQI(ecg_cleaned, sampling_rate=1000, window=1024, num_spectrum=[5, 15],  dem_spectrum=[5, 40], **kwargs):

    psd = signal_power(ecg_cleaned, sampling_rate=sampling_rate,
                       frequency_band=[num_spectrum, dem_spectrum],
                       method="welch", normalize=False, window=window, **kwargs)

    num_power = psd.iloc[0][0]
    dem_power = psd.iloc[0][1]

    return num_power / dem_power

 def _ecg_quality_basSQI(ecg_cleaned, sampling_rate=1000, window=1024, num_spectrum=[0, 1], dem_spectrum=[0, 40], **kwargs):
  
      psd = signal_power(ecg_cleaned, sampling_rate=sampling_rate,
                         frequency_band=[num_spectrum, dem_spectrum],
                         method="welch", normalize=False, window=window, **kwargs)
  
      num_power = psd.iloc[0][0]
      dem_power = psd.iloc[0][1]
  
      return 1 - num_power / dem_power

two functions in neurokit2.ecg.ecg_quality

how can use these functions individually??

Tam-Pham, 2021-08-22:
Hi @alexxony, 

I don't think I understand the issue that was raised. What do you mean by using them individually?

If possible, could you share with us the **full script** where you try to use these functions (from importing, reading data....)? A full script is one that we can run and replicate the exact error that you faced. For now, it's not clear what have you tried to do with these 2 functions

```python
def _ecg_quality_pSQI(ecg_cleaned, sampling_rate=1000, window=1024, num_spectrum=[5, 15],  dem_spectrum=[5, 40], **kwargs):

    psd = signal_power(ecg_cleaned, sampling_rate=sampling_rate,
                       frequency_band=[num_spectrum, dem_spectrum],
                       method="welch", normalize=False, window=window, **kwargs)

    num_power = psd.iloc[0][0]
    dem_power = psd.iloc[0][1]

    return num_power / dem_power
```

```python
 def _ecg_quality_basSQI(ecg_cleaned, sampling_rate=1000, window=1024, num_spectrum=[0, 1], dem_spectrum=[0, 40], **kwargs):
  
      psd = signal_power(ecg_cleaned, sampling_rate=sampling_rate,
                         frequency_band=[num_spectrum, dem_spectrum],
                         method="welch", normalize=False, window=window, **kwargs)
  
      num_power = psd.iloc[0][0]
      dem_power = psd.iloc[0][1]
  
      return 1 - num_power / dem_power
```

DominiqueMakowski, 2022-09-21:
Hopefully this is addressed but feel free to re-open if not

closed 2022-09-21 by DominiqueMakowski