neuropsychology/NeuroKit #500: HRV: SDANN and SDNNI

DominiqueMakowski, 2021-08-21:
Regarding this: https://github.com/neuropsychology/NeuroKit/issues/131

SDANN (average) and SDNNI (index) are technically defined as the average or SDs, or the SD of averages **"of 5min windows within a period of 24h"**. Since these are relatively arbitrary values, I suggest we name these indices **SDANN5** and **SDNNI5**, to explicitly mention the 5min windows. They can be computed on any signal which length is > to 15min but, obviously, it is more meaningful with very long signals. 

That said, based on the same principle of using time windows' values to capture more long-term changes, and/or having more robust indices, we could add variants, like **SDANN1**, **SDANN2** (1 min or 2min windows), which could be used on much shorter signals.

Aside from the benefit in terms of robustness, the question of their relevance is also conceptual, whether there is any information (i.e., physiological mechanisms acting at this scale) within 1 or 2 min windows that is not captured by the other short-term indices.

see https://www.ncbi.nlm.nih.gov/pmc/articles/PMC5624990/ and https://www.jacc.org/doi/10.1016/j.jacc.2006.06.042

Tam-Pham, 2021-08-22:
So do you suggest we add SDANN5 and SDNNI5, together with the ones that work with 1min and 2min windows?  And only return if the signal length is 3x the required window length (e.g., 15 mins for SDANN5)?

DominiqueMakowski, 2021-08-22:
yeah that's what to me would make kinda sense. The only question is wether SDANN1/2 and SDNNI1/2 are worth a try, and why not SDANN3, 4, 6 etc. What do you think

Tam-Pham, 2021-08-23:
Well, for resting-state data, I don't think there is any data that is longer than say 10 mins and so having SDANN/SDNNI for windows of 1 minute and 2 minutes seems to make the most sense. We include the 5 mins windows since it's corresponding to the 24hour guideline. I don't know having such indices for such short windows would capture any worthy variability. But then we can include them anyway for exploratory and since having these indices for the 5-minute window would allow the hrv_time() to be complete.

closed 2021-08-24 by Tam-Pham