neuropsychology/NeuroKit #506: ecg_delineate DWT wrong on HR 140

nachoXXL, 2021-08-24:
**Describe the bug**
When using ecg_delineate with method = 'dwt' on a simulated ECG with hr=140, sampling_rate=500, the events in the resulting image seem all wrong. Changing the sampling_rate has no effect.

**To Reproduce**
In a jupyter notebook:
import neurokit2 as nk
plt.rcParams['figure.figsize'] = [25, 5]
sr = 500
hr = 140

```
ecg_signal = nk.ecg_simulate(duration=10, noise=0.0, heart_rate=hr, sampling_rate=sr)
_, rpeaks = nk.ecg_peaks(ecg_signal, sampling_rate=sr)
_, waves_dwt = nk.ecg_delineate(ecg_signal, rpeaks, sampling_rate=sr, method="dwt", show=True, show_type='all')
```

**Screenshots**
[image]

**System Specifications**
```
- OS: Linux ( 64bit) 
- Python: 3.7.11 
- NeuroKit2: 0.1.4 

- NumPy: 1.21.0 
- Pandas: 1.2.5 
- SciPy: 1.7.0 
- sklearn: 0.24.2 
- matplotlib: 3.4.2

```

Tam-Pham, 2021-08-24:
hi @nachoXXL 

Thank you for raising the issue. I can replicate the problem and before looking at it in detail, I suspect the edge case is due to a very HR of the simulated signal. We were tunning the algorithms using signals of normal rate ranges (mostly at rest or non-strenuous activity). 

Do give me some time to look into the details and rule out any plotting problems and confirm that the current implementation is inaccurate with too high/too low HR. Potentially, if that's the case, I will look into tunning the search windows according to the rate of the signals.

nachoXXL, 2021-08-24:
Hi Tam-Pham,
It does not seem to be a plotting problem. If I analyse the results of ecg_delineate the data is already wrong. The events seem to shift. Some examples. Note I focus on the on/offset event as the peaks seems okay.

```
hr = 60
sr = 1000

ecg_signal = nk.ecg_simulate(duration=5, noise=0.0, heart_rate=hr, sampling_rate=sr)
_, rpeaks = nk.ecg_peaks(ecg_signal, sampling_rate=sr)
_, waves_dwt = nk.ecg_delineate(ecg_signal, rpeaks, sampling_rate=sr, method="dwt", show=False, show_type='all')
d_color = dict(P='red', Q='orange', R='green', S='orange', T='blue')

plt.plot(ecg_signal)

for ky, waves in waves_dwt.items():
    for ndx,v in enumerate(waves):
        if ky.endswith('sets'):
            plt.axvline(v, color=d_color[ky[4:5]], linestyle='--')
            plt.text(v+3, 1.2, f'{ky[4:5]}{ndx}')
```
Gives:
[image]

However with:
```
hr = 60
sr = 1000

ecg_signal = nk.ecg_simulate(****, noise=0.0, heart_rate=hr, sampling_rate=sr)
```
I get:
[image]

And warnings about (numpy) nan's.

Hope this helps.

Tam-Pham, 2021-08-24:
Thanks much for showing me your examples. May I know what is the code for your second plot? Is it of a different hr?

nachoXXL, 2021-08-24:
Hi Tam-Pham,
Sorry for the copy-paste error. The second image is created with hr=140.

Tam-Pham, 2021-08-25:
No worries 👍 

It seems clear that we will need to work on optimizing the function for different HR. I will follow up on this issue as soon as I can.

Tam-Pham, 2021-09-03:
Hi @nachoXXL,

I have recently added a function in `nk.ecg_delineate()` to allow the parameters to be adjusted by the detected HR of the signal
https://github.com/neuropsychology/NeuroKit/blob/9725a6f7fcc31bbab01096c6758a87e58d74530a/neurokit2/ecg/ecg_delineate.py#L241

Specifically, the parameters that are being adjusted here are the degree of the discrete wavelets and the search windows for the different features. These are the chosen parameters because naturally when HR increases, we would expect the search windows to shrink accordingly. 

And so for the simulated ECG with HR of 120:
```python
import neurokit2 as nk
# Simulated
ecg_signal = nk.ecg_simulate(duration=10, noise=0.0, heart_rate=120, sampling_rate=1000)
cleaned = nk.ecg_clean(ecg_signal, sampling_rate=1000)
_, rpeaks = nk.ecg_peaks(cleaned, sampling_rate=1000)
signals, waves = nk.ecg_delineate(cleaned, rpeaks, sampling_rate=1000, show=True, show_type="all")

import matplotlib.pyplot as plt
d_color = dict(P='red', Q='orange', R='green', S='orange', T='blue')
fig= plt.figure()
plt.plot(ecg_signal)

for ky, wave in waves.items():
    for ndx,v in enumerate(wave):
        if ky.endswith('sets'):
            plt.axvline(v, color=d_color[ky[4:5]], linestyle='--')
            plt.text(v+3, 1.2, f'{ky[4:5]}{ndx}')
        elif ky.endswith('Peaks'):
            plt.axvline(v, color=d_color[ky[4:5]])
            plt.text(v+3, 1.2, f'{ky[4:5]}{ndx}')
```

[image]

You can see that most features are now being identified pretty accurately, except for P-offsets and R-onsets. Speculatively, it seems that these features get very flat as the HR increases in the simulated signals. I can definitely tune the parameters of the delineator further to pick up these small and flat features (become more sensitive when HR is higher). Nevertheless, I have decided not to do so as these changes that we observe in the simulated signals might not hold for real signals. And further tunning of the delineator might introduce more problems than what it can actually solve. Especially since previous settings of the delineators were specifically tested with real ECG signals. 

I have also tested the updated delineator with real ECG signals (resting-state):
```python
# real data
bio_data = nk.data("bio_resting_8min_100hz.csv")
cleaned = nk.ecg_clean(bio_data["ECG"], sampling_rate=100)
_, rpeaks = nk.ecg_peaks(cleaned, sampling_rate=100)
signals, waves = nk.ecg_delineate(cleaned, rpeaks, sampling_rate=100, show=True, show_type="all")
```
[image]

If ECG signals of high HR are what you are working with, and you happen to have a few samples of those, please do feel free to share those with us and I can look into tuning with actual signals.

nachoXXL, 2021-09-03:
Hi Tam,
Nice job! Will give it a look when I've time. Give me a week or 2.
Kind regards,
Nacho

nachoXXL, 2021-09-20:
Hi Tam,
I'm back. However I can't find the change you mentioned. Not in latest nor in the distribution.
Can you give me an indication where it is please?

In the images you posted I see that the peaks are not the real peaks. I already noted that in my own
tests. Is that a known problem?

Kind regards,
Nacho

Tam-Pham, 2021-09-22:
Hey! Welcome back 😄 

I made the changes in #522. It's not merged to the dev branch yet.

> In the images you posted I see that the peaks are not the real peaks. I already noted that in my own tests. Is that a known problem?

Could you clarify this a little more? I'm afraid I'm not sure which peaks you are referring to 🤔

closed 2021-09-22 by nachoXXL