neuropsychology/NeuroKit #510: bug in hrv_nonlinear

richrobe, 2021-08-25:
The `_hrv_dfa()` function of the hrv_nonlinear module (NeuroKit version 0.1.4) throws an error for some R peak input data. The error occurs when trying to create the `long_window` array (line 469) because, at that point, `n_windows_long` is -1. The function input is being sanitized in the function, but only after the array creation.

```
short_window = np.linspace(dfa_windows[0][0], dfa_windows[0][1], n_windows_short).astype(int)
long_window = np.linspace(dfa_windows[1][0], int(max_beats), n_windows_long).astype(int)
```

This error could be solved by moving the creation of the `long_window` array into the else statement of the "sanitize max_beats" block. Then, the function would just issue a warning that DFA_alpha2 will not be calbulated, but now throw an error.

Tam-Pham, 2021-08-25:
hi @richrobe 

Many thanks for raising this issue. The most recent release is honestly quite buggy and we will try to release a patch version to fix the issues asap. Do let us know if you discover any other bugs while trying out!

richrobe, 2021-08-25:
Alright, will do! :-)

closed 2021-08-30 by Tam-Pham