neuropsychology/NeuroKit #517: Flexible for int8 and float64

DominiqueMakowski, 2021-08-29:
> ecg_peaks() can handle with int8 format ("uncleaned"), but ecg_delineate() cannot, and seems to accept only float64 (such as returned by ecg_clean()). However, with float64 datasets it quickly runs into memory issues.

Is this something that we can / want to fix?

### Discussed in https://github.com/neuropsychology/NeuroKit/discussions/516

<div type='discussions-op-text'>

<sup>Originally posted by **J4NPM** August 30, 2021</sup>
Hi everyone,

I am having issues when dealing with large ECG datasets in neurokit2. 
ecg_peaks() can handle with int8 format ("uncleaned"), but ecg_delineate() cannot, and seems to accept only float64 (such as returned by ecg_clean()). However, with float64 datasets it quickly runs into memory issues.
I also tried running ecg_delineate() on single  heartbeats (such as the epochs returned by ecg_segment() or manually selecting small segments of data) and that also raised errors. 
How to handle large datasets?
Is there a way to work entirely with int8 datasets?
Is there a way to use ecg_delineate() on a single beat?

Thanks for any help!
</div>

zen-juen, 2021-08-30:
This is probably due to the later step of sanitizing the columns in `epochs_create` (the int8 signal is detected as an object and then rendered as a string)

https://github.com/neuropsychology/NeuroKit/blob/abfd08db61d9d9142d796fd2efac149e8f4636e7/neurokit2/epochs/epochs_create.py#L162-L176

Even though this can be easily fixed/avoided, indeed I'm not sure if we may encounter more problems down the pipeline if int formats are accommodated 🤔

DominiqueMakowski, 2021-09-12:
@Tam-Pham *ping*

closed 2022-09-27 by stale[bot]