neuropsychology/NeuroKit #524: emg_process() activation offset on last element bug

Vasco-Cardoso, 2021-08-30:
Hello!

I seem to have found a small bug upon running the neurokit2 _emg_process()_ function with a data structure where the signal's activation offset is the last element of the structure, it returned error saying that the element wasn't on the structure.

I managed to eliminate this error by making a simple adjustment on **neurokit2/signal/signal_formatpeaks.py line 67** as the following image shows:

[image]

With this change i managed to run the function to my file with no problems.

The versions my system is running are:
- neurokit2==0.1.2
- tensorflow==2.4.1
- pandas==1.2.2

zen-juen, 2021-09-01:
Hi @Vasco-Cardoso, indeed it is important to ensure that the detected emg indices are within the length of the signal. Thanks for pointing that out and proposing a potential solution! However, I think the issue here is more to do with making sure that the extracted onsets/offsets (from `emg_activation`) are within the signal before letting `signal_formatpeaks` sanitize them - so I'd say it's best to leave the latter untouched.

https://github.com/neuropsychology/NeuroKit/blob/c1104386655724a5c624e740abf651c939fc5e48/neurokit2/emg/emg_activation.py#L370-L382

@DominiqueMakowski  We can make a small modification to enforce this:

```
    # make sure indices are within length of signal
    onsets = np.array([i for i in activations["onset"][valid] if i < len(activity)])
    offsets = np.array([i for i in activations["offset"][valid] if i < len(activity)])
```

DominiqueMakowski, 2021-09-01:
can't we simply do 
```python
# Cross-comparison 
valid = np.isin(activations["onset"], baseline["offset"]) 
onsets = activations["onset"][valid]
offsets = activations["offset"][valid] 

# make sure indices are within length of signal
onsets = onsets[onsets < len(activity)]
offsets = offsets[offsets < len(activity)]
```

But how come it's possible to have onsets and offsets bigger than the length 🤔 ?

zen-juen, 2021-09-01:
This probably won't be a problem for onsets actually, just offsets - since in our code offset indices are derived by `activations["onset"] + activations["duration"]` so in this edge case here where the activated portions (durations) end exactly at the last data point, the offset then becomes detected as `len(signal) + 1` index

And yes you're right @DominiqueMakowski `onsets = onsets[onsets < len(activity)]` is better 😆

closed 2021-09-03 by zen-juen