neuropsychology/NeuroKit #528: Confusion between exhalation peaks/troughs and inhale/exhale onsets [RSP]

shanelindsay, 2021-09-01:
Neurokit labels peaks in the respiration signal as inhalation peaks and troughs as exhalation troughs (e.g. the rsp_plot graph). The example https://neurokit2.readthedocs.io/en/latest/examples/rrv.html doesn't specify the nature of the respiration measurment (but perhaps should), though here refers to inspiration peaks and exhalation troughs. 
This had potential for confusion, as peaks and troughs in a respiration signal have different meaning depending on the type of recording and the measure (i.e. volume, spirometer or breathing belt). 
The code which calculates whether a sample is inspiration or expiration assumes that a trough is the start of inhalation, which would the case for breathing belt data (and I assume spirometer users might not be using neurokit for analysis). 

However, some of the code (if I am reading it right) seems to assume that "peaks" are inhalation onsets. For example, 
https://neurokit2.readthedocs.io/en/latest/_modules/neurokit2/rsp/rsp_rrv.html#rsp_rrv
bbi = np.peaks(peaks) / sampling rate * 1000
Here bbi is breath to interval, which is used to calculate a number of measures. 
Traditionally these measures would use the onset of the inhalation as the start of a breath cycle not the start of the exhale (i.e. the "inspiration peak" in neurokit terminology), so it seems the code should be **np.peaks(troughs)**. 
So this suggests a few possible changes - fixing the code to use troughs not peaks (and I think a few other functions are affected), being more explicit about the respiration recording type assumed in the docs, and perhaps shifting more to labelling peaks and troughs as inhale onsets and exhale onsets instead of inspiration/exhalation troughs and inspiration peaks (which would reduce ambiguity of the meaning of a "inspiration/exhalation peak/trough").
[note, some functions might work very similar regardless of using peaks and troughs, but in terms of reporting of measures inhale onsets would be better to fit with standard methods)

zen-juen, 2021-09-02:
Great call! Interesting that the shading of our rsp plot is actually in line with what was raised here, in that the trough is the start of the breath cycle (and hence inhalation onset), although our operationalization of bbi as `np.diff(peaks)` implies that each peak is the inhalation onset. 
[image]

And so if I'm understanding correctly, here the labels of exhalation troughs should be inhalation onsets and inhalation peaks should be exhalation onsets. Adding this to our [general roadmap](https://github.com/neuropsychology/NeuroKit/projects/4)!

shanelindsay, 2021-09-02:
Also to add a separate but related point:

In this code https://neurokit2.readthedocs.io/en/latest/_modules/neurokit2/rsp/rsp_findpeaks.html :
```
_rsp_findpeaks_extrema(rsp_cleaned):
    # Detect zero crossings (note that these are zero crossings in the raw
    # signal, not in its gradient).
    greater = rsp_cleaned > 0
```
I think it might be based on khodadad2018 ?

For RIP measurements are often in arbitrary units where zero has no meaning. So the raw signal might not have a zero, unless it is z-transformed. This differs from other methods like spirometry and electrical impedance tomography where 0 crossing has a specific meaning e.g. in spirometry downwards zero crossing marks exhale onset. So I haven't thought through the consequences of this code but raising it as a potential issue.

zen-juen, 2021-09-13:
Hi @shanelindsay, in the previous PR (ref above) I've changed all instances of "inhalation peaks" and "exhalation troughs", to "peaks (exhalation onsets)" and "troughs (inhalation onsets)" respectively in the docstrings. The explicit labelling of onsets is done only in `rsp_plot` and I left the namings in the generated signal columns/dictionary info as they are (i.e., `RSP_Peaks` and `RSP_Troughs`) since it seems more consistent with our API like that. `rsp_rrv()` also now correctly uses troughs rather than peaks. 

However, I have one question regarding the computing of respiratory rate - as you can see in our 
[code](
https://neurokit2.readthedocs.io/en/latest/_modules/neurokit2/rsp/rsp_rate.html), there are currently two methods to do so:
https://github.com/neuropsychology/NeuroKit/blob/c1104386655724a5c624e740abf651c939fc5e48/neurokit2/rsp/rsp_rate.py#L24-L28
The latter is independent of the indices but for the former, does it matter whether periods are computed from peaks or troughs?

zen-juen, 2021-09-14:
Regarding the above-mentioned issue about respiratory rate, see https://www.mdpi.com/1424-8220/20/6/1607 who concurs with the use of peaks from belt measurements, but with an additional threshold implemented:

> The respiration rate was calculated by counting the peaks from the respiration belt, which were used as the ground
truth. The threshold for detecting peaks in the respiration belt data was calculated using Equation (9). The term RGnd(n) is the signal measured from the respiration belt. If the peak was larger than the threshold, it was detected, and the peak was counted only when the interval between the peaks was 2.5 s or more.
Threshold = 0.8 × (1/(max(RGnd(n)) − min(RGnd(n)))) (9)

shanelindsay, 2021-09-14:
Great to see those changes.

In answer to your question, in one sense it might not matter, since you are
going to get similar results either way. But I think it does matter in a
couple of ways. In that paper you cite, they only care about respiration
rate, nothing else. And it is primarily about using radar to detect
respiration rate, and maybe peaks are better here for that method.

In neurokit, the respiration analysis emcompasses lots of aspects of the
respiration signal.

So it seems a bit weird (and a reviewer might think it weird) in a paper to
say we calculated the start of the breath as inspiration onset (for various
metrics), but for calculation of respiratory rate we counted exhalation
onset.

The code seems to suggest it would be straightforward to change peaks to
troughs, so this is what I would advocate for.

zen-juen, 2021-09-15:
Yes it's indeed true that many studies pit more modern/rare respiration measurement techniques (like radar) against the respiratory signal as measured by a belt as a reference. Based on a breadth-focused search, I haven't yet found any papers that are purely focused on evaluating the latter.

I also agree that it is concerning if our operationalization of BBI is inconsistent with the way we compute respiratory rate, as you correctly highlighted above. Given that respiratory rate is akin to the number of respiration events in time, it makes sense to compute it based on periods between successive inspirations (inspiration onset-to-inspiration onset) - which this [paper](https://annals-csis.org/Volume_8/pliks/329.pdf) also does. What do you think? @DominiqueMakowski

DominiqueMakowski, 2021-09-15:
i.e. based on troughs?

zen-juen, 2021-09-15:
> i.e. based on troughs?

Yes 😄

closed 2021-09-23 by zen-juen