neuropsychology/NeuroKit #529: Implementation of additional RSP breath detection algorithms - RSP

shanelindsay, 2021-09-02:
There are lots of methods for respiration analysis of breathing belt/RIP signals, but two good ones that could be part of neurokit  might be these two, which could be better defaults than the current one:

## 1. Breathmetrics (some code for signal generation is already implemented in neurokit)

Breathmetrics (open source in MATLAB) is designed for spirometry and athe lgorithm well validated but has adaptations for RIP described in the paper though currently not validated. 

One major advantage is it can encode pauses, which other algorithms do not. To integrate this in NK would involve some additional work.

Another advantage is that
[open source Python code based on Breathmetrics designed for RIP signals exists here](https://mwlodarczak.gitlab.io/RespInPeace/_modules/rip.html#Resp.extract_features) - which is described in this short paper [RespInPeace: Toolkit for processing respiratory belt data](https://zenodo.org/record/3246019) which would make integrating with NK much easier. This also has some useful functions such as "Calibrate respiratory signal in absolute units of volume given a vital capacity (VC) manoeuvre."

Integration would also allow better estimation of inhalation and exhalation times (due to detecting pauses) along with tidal volume and their ratios (Ti/Ttot, Vt/Ti) which are important for respiratory analyses, which would significantly enhance the measures available in neurokit. 

## 2. [Harrison et al. 2021](https://www.sciencedirect.com/science/article/pii/S1053811921000641) "A Hilbert-based method for processing respiratory timeseries"

This is mentioned in this [issue](https://github.com/neuropsychology/NeuroKit/issues/495) in the context of calculating RVT, but the suggestion here is to use it for breath detection along with rate and amplitude. While not validated it appears quite sophisticated and designed for RIP measurement.  

[Open source MATLAB code here from PhysIO toolbox] (https://github.com/translationalneuromodeling/tapas/blob/master/PhysIO/code/model/tapas_physio_rvt_hilbert.m)

PhysIO also has some code for respiratory processing not using the Hilbert method

DominiqueMakowski, 2021-09-02:
Thanks for all this info! So currently the main RSP-processing-related functions are the following:

https://github.com/neuropsychology/NeuroKit/blob/c1104386655724a5c624e740abf651c939fc5e48/neurokit2/rsp/rsp_process.py#L67-L77

i.e., cleaning, peak detection, and phase - amplitude. 

Currently, these functions have usually a `method` argument (e.g. `khodadad2018`). However, as you mentioned, these "methods" were usually developed for given techniques, and for now we kinda treated them as if they were the same. 

So as far as I understand, the steps for improving the RSP features would be to 1) add more methods specific to other techniques (and more relevant to rsp belt as I *think* it's what many NK users use).  2) I wonder whether then it would interesting to add in the main `rsp_process()` function a `technique` argument, set by default to "RIP" or "belt". This meta-argument could then modify the `method` arguments accordingly... or maybe it's just easier to have only the `method` argument and simply emphasize in the docs which methods correspond to what technique. We would need to think about the exact API a bit more, but the first step is to identify what functions need to be changed and what needs to be added.

Now practically speaking as much as we would like to improve this aspect of NK (and it would potentially benefit a lot of users), our very smol team is currently focused on developing other features, so we won't be able to get into that anytime soon. We would be super happy to have you onboard if you were interested in helping out with some implementations :) naturally, we'll help and assist as much as we can! Let us know

shanelindsay, 2021-09-07:
Yes I think having an argument would be good - at the least it would make it explicit rather than implicit (see other issue I raised).

Thinking about it, then it seems like the difference in measuring devices is similar in some ways to the difference of assessing heartbeats with, say, PPG vs. ECG, which have their own functions. This suggests to be that the technique should be earlier in the workflow. I imagine in a future more fully fledged respiration processing tool will need to differentiate not just method but the measure.

So there are a class of methods which estimate chest and abdomen movements, such as RIP, accelerometers, video, radar, etc...with these methods you can infer respiratory flow and/or volume (which could have associated neurokit functions for transformation, e.g. taking the derivative), and there are other techniques which measure flow/volume more directly, and then you might have functions that work on this kind of data. So it seems you might want `measure` `technique` and `method`.

In the shorter term I think improving the breathing belt workflow (on the assumption of biggest demand) would be worthwhile - if we make any progress with this (i.e. implementing some other techniques for breath detection) I will let you know.

closed 2022-12-03 by stale[bot]