neuropsychology/NeuroKit #557: Standardize and tidy up finding the elbow/knee/plateau point

DominiqueMakowski, 2021-10-09:
Finding the elbow of a curve is a very common problem and we use it in multiple instances. Currently we always have bespoke implementations but we could gather them into a unique `find_plateau/elbow()` function that would help in terms of re-usability, maintainability and testing.

zen-juen, 2021-10-13:
I thought of a few ways to implement this, trial-and-errored them, and I think the best I can think of for now is like so (see new `find_plateau()` [function](https://github.com/neuropsychology/NeuroKit/blob/feature/complexity_multichannel_data/neurokit2/misc/find_plateau.py) in the case of an increasing plateau)

1. extract indices in increasing segments where their gradients are becoming less positive (i.e. decelerating)
2. troughs like in the figure below may be detected from (1) so filter out these inverse peaks
3. from the filtered indices of (1) + (2), find the top 10% of indices with the greatest deceleration of gradients
4. if after step 3, there's more than 1 optimal index, extract the one which has the higher magnitude 

[image]

Perhaps there may be a more straightforward way to do this 🤔

closed 2022-09-20 by stale[bot]

DominiqueMakowski, 2022-09-21:
This exists now as [find_knee()](https://neuropsychology.github.io/NeuroKit/functions/misc.html?highlight=find_knee#neurokit2.misc.find_knee)