neuropsychology/NeuroKit #558: ECG peak detection robust to signal inversion

raphaelvallat, 2021-10-12:
**Question and context**

Hi here! 

Great work on the package 👏 👏  

I am analyzing a large database of ECG recordings (n=2000+) and I suspect that for some of those (but not all), the ECG data is inverted. I do not have the time to do a manual check of all the files. My question is therefore whether there is a preferred method that I should use in `ecg_clean` and `ecg_peaks` that will be robust to such polarity inversion. I have been using the default method ("neurokit") and it seems to be working quite well, but maybe there's a better alternative.

PS: I think this is related to https://github.com/neuropsychology/NeuroKit/issues/368.

Thanks for your help!
Raphael

DominiqueMakowski, 2022-09-21:
Did you have any luck with that in the end?

raphaelvallat, 2022-09-21:
Nope, ended up working on other projects and never had the time to finish this. Feel free to close the issue!

DominiqueMakowski, 2022-10-03:
FYI @raphaelvallat, @danibene implemented a new `ecg_invert()` function that detects whether inversion is needed (and inverts it if need be). Closing this then :)

closed 2022-10-03 by DominiqueMakowski