neuropsychology/NeuroKit #572: TypeError: unsupported operand type(s)

nieshr, 2021-11-05:
I got a TypeError when using the nk.ecg_delineate() function and I guessing it's the Python's version problem? How can I fix this?

The code snippet is like this:
_, waves_peak = nk.ecg_delineate(ecg_data, rpeaks_data, 500, method="peak")

which produces this message when it running:

neurokit2\ecg\ecg_delineate.py", line 813, in _ecg_delineator_peak_Q
    Q = signal_findpeaks(-1 * segment["Signal"], height_min=0.05 * (segment["Signal"].max() - segment["Signal"].min()))
TypeError: unsupported operand type(s) for -: 'str' and 'str'

The system I am using
```
nk.version()
- OS: Windows (WindowsPE 64bit)
- Python: 3.8.5
- NeuroKit2: 0.1.4.1

- NumPy: 1.18.5
- Pandas: 1.3.4
- SciPy: 1.4.1
- sklearn: 1.0.1
- matplotlib: 3.4.3
```

nieshr, 2021-11-05:
I convert the data in ecg_data[] to float manually and solve the problem.

DominiqueMakowski, 2021-11-06:
Yes, the ECG signal must be made of numbers indeed :)

closed 2021-11-06 by DominiqueMakowski