neuropsychology/NeuroKit #583: Reading in ECG Data collected from custom build ECG device

SolaimanR, 2021-11-21:
In order to prepare a script that reads data from a custom build device, we are trying to use neurokit2. When going through documentation and using the provided sample datasets, we can easily read and work with the data. However, trying to load sample data from other sources fails. 

For example, neurokit2 struggles with sample data from a different source (attached File 1). We read in the data as follows:

```import neurokit2 as nk
import numpy as np
import pandas as pd
import matplotlib.pyplot as plt
import seaborn as sns
%matplotlib inline
plt.rcParams['figure.figsize'] = [8, 5]

#File 1 Sample Data
columnname = ['ECG']
ecg_signal =  np.fromfile("file1.csv",dtype=float)
ecg_signal = pd.Series(ecg_signal, name='ECG')

_, rpeaks = nk.ecg_peaks(ecg_signal, sampling_rate=250)
plot = nk.events_plot(rpeaks['ECG_R_Peaks'], ecg_signal)
```

We also tried data from Physionet. We saw a script that downloads and prepares those files for neurokit2. But since we are trying to prepare the script for our own data (which will be available to us next month), we wanted to simply read in the csv files that we created from the .dat files from physionet (attached file2). 

What exactlty is the difference between the Neurokit2 sample data "ecg_3000hz" and the Physionet .dat files or the file1. We took the sampling rate under consideration but other than that, the actual file format is identical, only the values differ, which shouldnt be a problem right? 

Here is how we read in the Physionet data:

```ecg_csv = pd.read_csv("file2.csv", header=None)
ecg_csv = ecg_csv.rename(columns={0:"ECG"})
ecg_series = pd.Series()
ecg_signal = ecg_csv["ECG"]

_, rpeaks = nk.ecg_peaks(ecg_signal, sampling_rate=360)
plot = nk.events_plot(rpeaks['ECG_R_Peaks'], ecg_signal)
```

Thank you for your help!

[file1.csv](https://github.com/neuropsychology/NeuroKit/files/7577163/file1.csv)
[file2.csv](https://github.com/neuropsychology/NeuroKit/files/7577217/file2.csv)

anshu-97, 2021-11-23:
Hi @SolaimanR, 

Just wanted to ask if there is any particular reason you chose to use `np.fromfile` instead of `pd.read_csv`?

Here's how the signal plot looks like when we import the file using `pd.read_csv` :
[image]

And here's how it looks like when we import it using `np.fromfile`:
[image]

I suspect this is because `np.fromfile` is meant to construct an array from text or binary data. 

As you can see here, this is the output of the imported signal using `np.fromfile`, which appears distorted from your original values:
```
array([3.22524129e-086, 1.72384437e-259, 3.22524126e-086, ...,
       1.72417062e-259, 1.57586916e-052, 1.96805813e-259])
```

whereas this is the output if we use `pd.read_csv` instead, which corresponds to the exact values of your file1.csv:
```
array([0.455, 0.46 , 0.45 , ..., 0.565, 0.58 , 0.585])

```

---
With regards to file2.csv, using a quick inspection, the values seem exceptionally small such that many of them are virtually treated as 0. May we ask how you converted the physionet files into your file2.csv? Perhaps you could share with us your script so that we can try investigating this issue further?

SolaimanR, 2021-11-23:
Thank you! I dont remember why I used numpy to read the data, but yes it works using pandas! As to file2, the way I converted the .dat files with the ECG Signal from Physionet did not work and I did not notice. I know converted the .dat files using a script I found online, which converts all the .dat files to .csv and it works like a charm using NeuroKit now!

[This is the script I found online to pull the .dat files as .csv files.](https://github.com/abhilampard/Physionet-CSV-Conversion/blob/master/script.py)

Thank you for your quick response and help.

anshu-97, 2021-11-24:
Hi @MoentyPython,

No problem! If there is nothing further at this time, I will go ahead and close this issue!