neuropsychology/NeuroKit #584: Peaks detection Failed

purushottam22, 2021-11-24:
I was trying detect all peaks ( P,Q, R, S, T) on an ecg data using this code 
:-
_, waves_peak = nk.ecg_delineate(ecg_signal, rpeaks, sampling_rate=3000, method="peak")

I encountered this error :-

--> 831     Q = signal_findpeaks(-1 * segment["Signal"], height_min=0.05 * (segment["Signal"].max() - segment["Signal"].min()))
    832     if len(Q["Peaks"]) == 0:
    833         return np.nan, None

TypeError: unsupported operand type(s) for -: 'str' and 'str'

R- peaks detection was successful.

zen-juen, 2021-11-29:
Hi @purushottam22 could you provide a sample of your ECG data so that we can try to reproduce the error for investigation? 😄

opsb, 2022-02-14:
Hey @zen-juen I ran into this issue while exploring the `ecg_delineate.ipynb` example. The repo at https://github.com/opsb/neurokit_examples is just the neurokit examples with poetry used for the deps. Using the deps pinned by poetry I saw the same error when running `ecg_delineate.ipynb` (the "Peak method" cell)

opsb, 2022-02-15:
It appears the issue is introduced at https://github.com/neuropsychology/NeuroKit/blob/master/neurokit2/epochs/epochs_create.py#L175. If I remove the sanitisation step then the ecg_delineate.ipynb example runs fine. Neither `string` nor `int64` seem like appropriate datatypes here so I'm not sure what this sanitisation step is there for.

DominiqueMakowski, 2022-02-15:
@opsb  could you try with the latest version (0.1.6)?

opsb, 2022-02-15:
Hey @DominiqueMakowski was trying to work out how I missed that but see it's just been released. I'm happy to report that the ecg_delineate.ipynb example runs fine against 0.1.6 :tada: I've updated https://github.com/opsb/neurokit_examples in case it's helpful to someone to have the dependencies pinned. The ecg_delineate looks great by the way, later on I'll try running some ECGs I've collected from apple watches through it. Many thanks for publishing the update!

closed 2022-10-16 by DominiqueMakowski