neuropsychology/NeuroKit #586: ValueError: x must be strictly increasing sequence. During ecg_process

ZamboniM, 2021-11-27:
**Describption**
I'm analyzing a dataset of ecgs, I can process 165 of them correctly, but 1 of them raise an error (I'll copy it down), the stacktrace is not very informative for me, but it seems like this is a bug in the library and not a problem with the data

**To Reproduce**

`raw_ecg = pd.read_csv("bug_ecg.csv")`
`nk.ecg_process(raw_ecg, sampling_rate=300)`

[bug_ecg.csv](https://github.com/neuropsychology/NeuroKit/files/7611927/bug_ecg.csv)

**System Specifications**
I tried both on Google Colab and my laptop with slightly different versions of the libraries

(Colab)
- OS: Linux ( 64bit) 
- Python: 3.7.12 
- NeuroKit2: 0.1.5 

- NumPy: 1.19.5 
- Pandas: 1.1.5 
- SciPy: 1.4.1 
- sklearn: 1.0.1 
- matplotlib: 3.2.2

(Laptop)
- OS: Linux (ELF 64bit) 
- Python: 3.9.7 
- NeuroKit2: 0.1.5 
- NumPy: 1.21.4 
- Pandas: 1.3.4 
- SciPy: 1.7.3 
- sklearn: 1.0.1 
- matplotlib: 3.5.0

**Output**

`Traceback (most recent call last):`
`  File "venv/lib/python3.9/site-packages/neurokit2/ecg/ecg_process.py", line 101, in ecg_process`
`    rate = signal_rate(rpeaks, sampling_rate=sampling_rate, desired_length=len(ecg_cleaned))`
`  File "venv/lib/python3.9/site-packages/neurokit2/signal/signal_rate.py", line 54, in signal_rate`
`    period = signal_period(peaks, sampling_rate, desired_length, interpolation_method)`
`  File "venv/lib/python3.9/site-packages/neurokit2/signal/signal_period.py", line 78, in signal_period`
`    period = signal_interpolate(peaks, period, x_new=np.arange(desired_length), method=interpolation_method)`
`  File "venv/lib/python3.9/site-packages/neurokit2/signal/signal_interpolate.py", line 74, in signal_interpolate`
`    interpolation_function = scipy.interpolate.PchipInterpolator(x_values, y_values, extrapolate=True)`
`  File "venv/lib/python3.9/site-packages/scipy/interpolate/_cubic.py", line 232, in __init__`
`    x, _, y, axis, _ = prepare_input(x, y, axis)`
`  File "venv/lib/python3.9/site-packages/scipy/interpolate/_cubic.py", line 61, in prepare_input`
`    raise ValueError("`x` must be strictly increasing sequence.")`
`ValueError: `x` must be strictly increasing sequence.`

zen-juen, 2021-11-29:
Hi @ZamboniM, it seems like the error is occurring at the computation of heart rate from R peaks. This is because there are two sample points in which the peaks are at the exact same location. Could you try `nk.ecg_process(bug_ecg, sampling_rate=300, method=pantompkins)` to change the method of finding peaks and see if this works for you?

ZamboniM, 2021-11-29:
Thanks @zen-juen it usually solves the issue, but on some ecgs I get the error reported in #532 . So I'm now trying to process with the default method and rollback to pantompkins when is needed, but I have 2 samples where both method "neurokit" and "pantompkins":
[bug_ecg_2.csv](https://github.com/neuropsychology/NeuroKit/files/7615699/bug_ecg_2.csv)
[bug_ecg_1.csv](https://github.com/neuropsychology/NeuroKit/files/7615700/bug_ecg_1.csv)

DominiqueMakowski, 2021-11-29:
I wonder whether we could tweak the peak detection method to prevent detecting identical indices @JanCBrammer

zen-juen, 2021-11-29:
For bug_ecg_1.csv, the signal appears to be ppg rather than ecg! You'll need to use `ppg_process` in that case (see [docs](https://neurokit2.readthedocs.io/en/latest/_modules/neurokit2/ppg/ppg_process.html) here) However, I just found a bug in `ppg_process`, do give me a few minutes to fix this and get back to you.

[image]

As for bug_ecg_2.csv, it appears that the error is because of this same issue of identical indices at  `ecg_peaks()` level in the pipeline. This seems to be when `correct_artifacts` is set to True here @DominiqueMakowski @JanCBrammer

Will open a PR for these two issues!

zen-juen, 2021-11-29:
@ZamboniM for the first sample data you can now install neurokit using the PR branch (as mentioned just above) and try:
```
ecg_1 = pd.read_csv('bug_ecg_1.csv', header=None)[0]
signals, info = nk.ppg_process(ecg_1, sampling_rate=300)
```

ZamboniM, 2021-11-29:
Thanks again @zen-juen. To me it looks more like an ecg than a ppg and it would very strange that a ppg signal ended up in the dataset.
Anyway I can ignore these two particular signals without problems and continue to work; so I'm ok

closed 2021-11-30 by zen-juen