neuropsychology/NeuroKit #589: Synchronous analysis of ECG & PPG

HotDog702, 2021-11-30:
If we have 2 (or more) signals, it will be good to analyze them in join. 
For example, if we have 1 or more ECGs + 1 or more PPGs, we can use known time relations between them to find peaks more accurately.

DominiqueMakowski, 2021-12-02:
It could be technically feasible to "cross-validate" heartbeats location in ``nk.bio_process()` that can take one ppg and one ecg together.

I'm not sure how "using known time relations between them" look like in practice... do you have any toy example or papers to show that?

vasileermicioi, 2021-12-29:
PTT - Pulse Transit Time is used to compute Blood Pressure, and it can be computed from sync ECG and PPG data

Papers (just googled to use as proof for the idea, not intended to recommend)
https://www.ncbi.nlm.nih.gov/pmc/articles/PMC4512231/
https://pubmed.ncbi.nlm.nih.gov/19697692/

Sensors capable to collect sync PPG and ECG
https://www.maximintegrated.com/en/products/interface/signal-integrity/MAX86150.html
https://ams.com/as7038rb#tab/description

the one from AMS is saying to compute blood pressure to meet IEEE Std 1708-2014 standard

danibene, 2022-11-01:
This reference uses consistency between ECG & PPG heart rates in determining signal quality: https://academic.oup.com/jamia/article/13/3/309/699677

HotDog702, 2022-11-16:
> This reference uses consistency between ECG & PPG heart rates in determining signal quality:
Good! 
But I have a little bigger idea: get "beats candidates" on ECG and PPG, then check quality of each signal, and leave only those, that have good quality. After that, basing on that "reliable" beat marks, try to find beats in noisy areas.
For example: using excellent ECG, you can split PPG to beats even if it has bad quality. Using good PPG you can select the correct R peaks among noise in ECG.

danibene, 2022-11-16:
I personally think that would be a great feature! Would you be able to/want to implement it?

closed 2023-02-05 by stale[bot]