neuropsychology/NeuroKit #591: Show plots

VegardIversen, 2021-12-03:
**Question and context**
Hi, 
Great library !
I have a question about this implementation.
```
ecg_signal = nk.data(dataset="ecg_3000hz")['ECG'] 
_, rpeaks = nk.ecg_peaks(ecg_signal, sampling_rate=3000) 
plot = nk.events_plot(rpeaks['ECG_R_Peaks'], ecg_signal)
```
This code will not show any plots, I can see why in the code, but is there a reason for why `plt.show()` is not called when `show=True`.
For me it would be logical to actually see the plot when i call a plotting function with `show=True` as a parameter, and not just return a Axes object.

zen-juen, 2021-12-04:
Hi @VegardIversen, it should definitely return a plot as you said, since the default is `show=True`. Here we call the pandas `plot()` function instead rather than matplotlib `plt.show()`. Can I check with you what python IDE you are using?

VegardIversen, 2021-12-04:
> Hi @VegardIversen, it should definitely return a plot as you said, since the default is `show=True`. Here we call the pandas `plot()` function instead rather than matplotlib `plt.show()`. Can I check with you what python IDE you are using?
@zen-juen 
I am using visual studio code. Im guessing the reason for it is that it seems to use jupyter notebook in the examples, and here you dont need to call the show function.

zen-juen, 2021-12-04:
Ah I think I misunderstood - I realize now that you're asking this more for conceptual understanding rather than asking for help to enable plotting!

Yes you're right about `plt.show()` being unnecessary when it comes to jupyer notebook or when using interactive mode (https://stackoverflow.com/questions/54422714/when-is-plt-show-required-to-show-a-plot-and-when-is-it-not) - I assume you're not using this in VS code? Are you running the code from the terminal?

VegardIversen, 2021-12-04:
Yes im running the code from terminal. 
Was just a question about if it should be like that.

zen-juen, 2021-12-04:
Hmm I didn't think it would be a problem since we have `return fig` which should act similar to `plt.show()`. Thoughts? @DominiqueMakowski

DominiqueMakowski, 2021-12-05:
My understanding of matplotlib's behaviour is too low to be of good advice here. I don't know what's the best practice...

From a design perspective, I think the goals would be, when running a function with `show=True`:

- The plot shows up
- Shows up only once (sometimes I've experienced the plot being duplicated)
- The plot can be "recovered" outside the function for further customization/processing

I'm not sure what's the standard way to achieve that. In any case, we should at least be consistent and adopt the same approach in all our plotting

zen-juen, 2021-12-07:
> I'm not sure what's the standard way to achieve that. In any case, we should at least be consistent and adopt the same approach in all our plotting

If I'm not wrong, we always initiate our matplotlib as `fig` and `return fig` at the end of the function (because sometimes we have internal functions that add more plotting features in between), which should make `plt.show()` not needed. But yeah good to double check

DominiqueMakowski, 2021-12-07:
> return fig at the end of the function (because sometimes we have internal functions that add more plotting features in between)

The `return fig` is only used if the function only does plotting, but for functions that return something else then we don't return the fig so I'm not sure how it works

VegardIversen, 2021-12-10:
I think it should be a plt.show() after initiating the plot. If you dont want unnecessary code when running it in jupyter you could make a check if its run in jupyter, or you can just include it either way because it will not affect jupyter (as far as i have found out).

closed 2022-09-22 by stale[bot]