neuropsychology/NeuroKit #594: Neurokit ecg_segment seems to have issues.

debugger98, 2021-12-21:
I'm trying to segment an ECG signal into heartbeats. Using `nk.ecg_segment` on cleaned waveform gives `r_peaks` as empty list. There are some list slicing in the following code, which generates this error. I'm working on PTBXL dataset, attaching the waveform file and python notebook for better understanding and debugging.

System Specifications

It's important that you give us some information about the system you are using. For that you can run:
- OS: Linux (ELF 64bit) 
- Python: 3.8.8 
- NeuroKit2: 0.1.5 

- NumPy: 1.19.5 
- Pandas: 1.3.3 
- SciPy: 1.5.4 
- sklearn: 0.24.2 
- matplotlib: 3.3.4

### Error stack
```
     13             ecg_cleaned = nk.ecg_clean(waveform, sampling_rate=sampling_rate)
     14 
---> 15         heartbeats = nk.ecg_segment(ecg_cleaned, sampling_rate=sampling_rate)
     16         return heartbeats
     17 

~/check_env/lib/python3.8/site-packages/neurokit2/ecg/ecg_segment.py in ecg_segment(ecg_cleaned, rpeaks, sampling_rate, show)
     56         rpeaks=rpeaks, sampling_rate=sampling_rate, desired_length=len(ecg_cleaned)
     57     )
---> 58     heartbeats = epochs_create(
     59         ecg_cleaned, rpeaks, sampling_rate=sampling_rate, epochs_start=epochs_start, epochs_end=epochs_end
     60     )

~/check_env/lib/python3.8/site-packages/neurokit2/epochs/epochs_create.py in epochs_create(data, events, sampling_rate, epochs_start, epochs_end, event_labels, event_conditions, baseline_correction)
    115     if epochs_end == "from_events":
    116         epochs_end = [i / sampling_rate for i in events["duration"]]
--> 117     parameters = listify(
    118         onset=event_onsets, label=event_labels, condition=event_conditions, start=epochs_start, end=epochs_end
    119     )

~/check_env/lib/python3.8/site-packages/neurokit2/misc/listify.py in listify(**kwargs)
     28     for key, value in args.items():
     29         if isinstance(value, list):
---> 30             args[key] = _multiply_list(value, maxi)
     31         else:
     32             args[key] = _multiply_list([value], maxi)

~/check_env/lib/python3.8/site-packages/neurokit2/misc/listify.py in _multiply_list(lst, length)
     36 
     37 def _multiply_list(lst, length):
---> 38     q, r = divmod(length, len(lst))
     39     return q * lst + lst[:r]

ZeroDivisionError: integer division or modulo by zero
```

[nk_error.zip](https://github.com/neuropsychology/NeuroKit/files/7757310/nk_error.zip)

zen-juen, 2021-12-28:
Hey @debugger98, using the script and data that you provided, I was able to obtain rpeaks. Is the waveform in the zip file the same dataset that you performed the script on?

Just one thing though, which is that the signal appears to be inversed
```
import neurokit2 as nk
import numpy as np
import pandas as pd

# 2nd lead of the waveform
waveform = np.load('15893-806179789.npz')['waveform'][1]

ecg_cleaned = nk.ecg_clean(waveform, sampling_rate=500)
ecg_cleaned_2 = ecg_cleaned * -1
pd.Series(ecg_cleaned).plot()
pd.Series(ecg_cleaned_2).plot()
```
[image]
[image]

```
_, rpeaks = nk.ecg_peaks(ecg_cleaned_2, sampling_rate=500)
print(rpeaks)
{'ECG_R_Peaks': array([ 414,  881, 1226, 1787, 2187, 2496, 2992, 3404, 3816, 4215, 4577,
       4858]), 'sampling_rate': 500}
```

debugger98, 2022-02-02:
I think it's good to go.

closed 2022-02-02 by debugger98